wled/WLED #2299: Support for Wemos ESP32-S2 mini

hutje, 2021-10-30:
Wemos has released a new controller.
https://www.wemos.cc/en/latest/s2/s2_mini.html
It would be really helpful if this controller would be compatible with WLED.
The USB-OTG makes it useful for easy flashing and can act as a USB-host and USB-client. 
The use of USB could make all kind of new connections possible in the future.

Aircoookie, 2021-10-30:
Nice little controller.
While the ESP32-S2 has not yet been tested enough to confirm that it fully works with all features of WLED, you can already build for it using the `esp32s2-saola` PIO environment.
Here is a link to a binary built for it: https://drive.google.com/file/d/1-yWHNwEvgG0cRIfjrD0LlQRm_Ezd-L0J/view?usp=sharing
, though you'll likely also need a bootloader for it, so I'd recommed compiling and uploading from VS code yourself.

Side note: We are already at 98.4% flash utilization with a 1.5MB app partition size on ESP32-S2. I don't know why the Espressif frameworks takes almost an extra 400kB compared to ESP32, but that means that OTA capability might not be sustainable on the S2 in the future.

mjoshd, 2021-12-08:
@Aircoookie 
I received a few of the Wemos S2 Mini modules in the mail recently and was able to get them working with both Tasmota and ESPEasy but I cannot get it to work with WLED (which was the main reason for purchasing them).

I was able to compile and upload without issue using VSCode and Platform IO but after joining it to my wifi and going to the ip address there are two errors.

The first one disappears too fast for me to read but the second one says `Error 10: Could not mount filesystem!`

[image]

After rebooting the device it goes back into AP mode.

Any ideas or next steps to try?

blazoncek, 2021-12-08:
This is a known bug, unresolved due to ESP core libraries used not supporting file system on S2.
You can try finding compatible LITTLEFS library and replacing it in platformio.ini

it-guy92, 2022-01-09:
Hey, 
i got the same Problem with "Error 10: Could not mount filesystem!" on my ESP32-S2 mini. Googled for replacing LITTLEFS but found nothing usefull. Maybe some experienced user can help?

blazoncek, 2022-09-15:
This is a copy of #966 
Closing.

Anyone interested may try https://github.com/blazoncek/WLED/tree/esp32-s2 
Please use this as PIO environment:
```ini
[env:lolin_s2_mini]
platform = espressif32@5.1.1
board = lolin_s2_mini
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
board_build.flash_mode = qio
upload_speed = 460800
upload_port = /dev/cu.usbmodem01
monitor_port = /dev/cu.usbmodem01
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags} ${esp32s2.build_flags}
  -DBOARD_HAS_PSRAM
  -D ARDUINO_USB_CDC_ON_BOOT
  -D WLED_USE_PSRAM
  -D WLED_WATCHDOG_TIMEOUT=0
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -D LEDPIN=16
  -D BTNPIN=0
  -D HW_PIN_SCL=35
  -D HW_PIN_SDA=33
  -D HW_PIN_CLOCKSPI=7
  -D HW_PIN_DATASPI=11
  -D HW_PIN_MISOSPI=9
  -D STATUSLED=15
  -D WLED_DEBUG
  -UWLED_USE_MY_CONFIG
lib_deps = ${esp32s2.lib_deps}
```

closed 2022-09-15 by blazoncek

mjoshd, 2022-09-16:
@blazoncek 

I compiled the repo you linked, along with the PIO settings you described, with no issues. Great work!

The device seems to run perfectly fine and I was able to save one preset without issue. 

I have not been able to save any additional presets though. After clicking the save button the device waits for a moment then seems to freeze and requires a reset, or power cycle, to become available again. 

I tried many of the effects and they all seemed to work perfectly.

I just wanted to test and provide some feedback. 

Thanks for your amazing efforts to make this board supported!

Let me know if you would like any other tests performed.

blazoncek, 2022-09-16:
I have it working ... kind of. With a lot of reboots or hangs that require flashing.
I upload presets created on another device as saving presets is error prone on ESP32 as well.
What I am seeing is reboots if I start a playlist that is switching presets.