wled/WLED #2301: Web server stops responding after effect applied, after some minutes reboots

pczekalski, 2021-10-31:
**Describe the bug**
AP stops responding after a while. Connecting in AP mode (directly to the esp8266), ESP is still generating the effect on the LEDs for some minutes, so I assume it is not dead. Interestingly, for a short time (1-2 minutes) there is an active WiFi connection but the webserver is dead (you cannot connect to the webserver nor over the web, nor with app), meanwhile scanning from the other devices does not show up WLED-AP at all. Some 5 minutes later it reboots, and I get WLED-AP present again (watchdog triggers reboot?)

**To Reproduce**
Connect to WLED-AP, change effect to the Aurora or any other, shuffle a bit with colour palettes (not necessary).

**Expected behavior**
WLED-AP should be visible and responding all time

**WLED version**
 - Board: NodeMCU ESP8266 (ESP 12E)
 - Version 1.0
 - Binary 0.12, ...esp8266.bin

Aircoookie, 2021-11-04:
Sounds like it might be a low memory condition.
How many LEDs are you running?
Is the ESP also connected to a network as client (dual mode) or is it AP-only?
You can try if 0.13.0-b4 improves the situation.

pczekalski, 2021-11-05:
Hi,

Thanks for the reply.
I've tested with 600 and 100 LEDs
ESP is in AP only mode.
I'll give it a try with the latest beta and let you know ASAP.

Regards

closed 2022-04-27 by stale[bot]