wled/WLED #2303: Bump della versione a 0.13.0-b4 "Toki" bug

cocolino83, 2021-10-31:
I noticed that there is a bug on version 0.13.0-b4 "Toki".  in practice, if you use all 10 LED outputs it is no longer possible to reach the favorite LED menu, the page no longer loads.  the ESP must be reset.  thanks a lot

Aircoookie, 2021-11-01:
Hi, thank you for the report. Can you send the output of `http://[WLED-IP]/cfg.json`? (don't worry, it doesn't include your wifi password)

cocolino83, 2021-11-08:
ciao scusa se ti rispondo adesso ma non ho avuto tempo. ti incollo comandi cfg.json. Come già detto alla giunta della decima uscita led quando si va su configurazione led sia da crome che da app la pagina non viene più aperta.

{"rev":[1,0],"vid":2109220,"id":{"mdns":"wled-6d5738","name":"WLED","inv":"Light"},"nw":{"ins":[{"ssid":"Vodafone-A40580074","pskl":10,"ip":[192,168,1,56],"gw":[192,168,1,1],"sn":[255,255,255,0]}]},"ap":{"ssid":"WLED-AP","pskl":8,"chan":1,"hide":0,"behav":0,"ip":[4,3,2,1]},"wifi":{"sleep":false},"hw":{"led":{"total":300,"maxpwr":850,"ledma":55,"rgbwm":3,"ins":[{"start":0,"len":30,"pin":[16],"order":0,"rev":false,"skip":0,"type":22},{"start":30,"len":30,"pin":[17],"order":0,"rev":false,"skip":0,"type":22},{"start":60,"len":30,"pin":[21],"order":0,"rev":false,"skip":0,"type":22},{"start":90,"len":30,"pin":[22],"order":0,"rev":false,"skip":0,"type":22},{"start":120,"len":30,"pin":[27],"order":0,"rev":false,"skip":0,"type":22},{"start":150,"len":30,"pin":[25],"order":0,"rev":false,"skip":0,"type":22},{"start":180,"len":30,"pin":[32],"order":0,"rev":false,"skip":0,"type":22},{"start":210,"len":30,"pin":[4],"order":0,"rev":false,"skip":0,"type":22},{"start":240,"len":30,"pin":[2],"order":0,"rev":false,"skip":0,"type":22},{"start":270,"len":30,"pin":[26],"order":0,"rev":false,"skip":0,"type":22}]},"btn":{"max":4,"ins":[{"type":2,"pin":[0],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]}],"tt":32,"mqtt":false},"ir":{"pin":-1,"type":0},"relay":{"pin":12,"rev":false}},"light":{"scale-bri":100,"pal-mode":0,"aseg":false,"gc":{"bri":1,"col":2.8},"tr":{"mode":true,"dur":7,"pal":0},"nl":{"mode":1,"dur":60,"tbri":0,"macro":0}},"def":{"ps":0,"on":true,"bri":128},"if":{"sync":{"port0":21324,"port1":65506,"recv":{"bri":true,"col":true,"fx":true,"grp":1},"send":{"dir":false,"btn":false,"va":false,"hue":true,"macro":false,"twice":false,"grp":1}},"nodes":{"list":true,"bcast":true},"live":{"en":true,"port":5568,"mc":false,"dmx":{"uni":1,"seqskip":false,"addr":1,"mode":4},"timeout":25,"maxbri":false,"no-gc":true,"offset":0},"va":{"alexa":false,"macros":[0,0]},"blynk":{"token":"","host":"blynk-cloud.com","port":80},"mqtt":{"en":false,"broker":"","port":1883,"user":"","pskl":0,"cid":"WLED-6d5738","topics":{"device":"wled/6d5738","group":"wled/all"}},"hue":{"en":false,"id":1,"iv":25,"recv":{"on":true,"bri":true,"col":true},"ip":[192,168,1,0]},"ntp":{"en":false,"host":"0.wled.pool.ntp.org","tz":0,"offset":0,"ampm":false,"ln":0,"lt":0}},"ol":{"clock":0,"cntdwn":false,"min":0,"max":29,"o12pix":0,"o5m":false,"osec":false},"timers":{"cntdwn":{"goal":[20,1,1,0,0,0],"macro":0},"ins":[]},"ota":{"lock":false,"lock-wifi":false,"pskl":7,"aota":true},"um":{}}

[cfg.zip](https://github.com/Aircoookie/WLED/files/7494161/cfg.zip)

Aircoookie, 2021-11-08:
Ah now I get the problem: The LED settings page values can not be constructed beyond 6-8 configured busses, since the 2k stack-allocated buffer is not sufficient. I measured that ESP32 still has 4k free stack with the 2k buffer, so increasing it to 3-4k would be no problem and likely fix this issue.

Fix coming soon!

closed 2021-11-09 by Aircoookie

cocolino83, 2021-11-09:
ciao. grazie per l'aiuto. ma non so come si fa ad aumentare il buffer. mi potresti dire come devo fare ad aumentarlo? grazie mille

PR #2323 (2021-11-09): Fix settings JS buffer too small
The previous buffer size was 2048b. This was not enough for the LED settings page with 10 busses.
This increases it to 3096b for ESP32 (issue should not be present on ESP8266 due to the 3 bus limit).
Measured free stack remaining before the change is 4k, after the change 3k, which should be enough leeway to not cause a stack overflow.

Likely fixes #2303