wled/WLED #2304: Skip First LED plus Reverse ends up leaving the last LED dark

timkingman, 2021-11-01:
**Describe the bug**
Turning on both Reverse (in LED setup) and Skip First LED ends up skipping the last (or Skip is ignored when Reverse is on?). If I turn on both, my sacrificial pixel (nearest) is animated (and the farthest pixel is dark, but this could just be my segment length?).

In writing this up, I think the right answer for me is to Skip First, don't Reverse in LED setup, and then Reverse Direction the single Segment 0. (OR I think I can turn on Reverse on the LED setup, turn off Skip First, and then just count my visible LEDs from the far end and don't count the sacrificial pixel which is now logically "last".) But I'll submit this anyway because Skip First sounds like it should always skip the nearest sacrificial pixel, my lights are mounted "reversed", and I don't think of my single string as having segments; and in case anybody else runs into my confusion too.

**To Reproduce**
I have a controller (NodeMCU), connecting immediately to a single sacrificial pixel, then about 5m of wire, then a run of 141 bulbs across my house. When facing the house, the controller is attached to the right end and the data arrows on the bulbs flow from right to left. I assume this is logically backwards, so I checked the Reverse box, in addition to Skip First LED because I have a sacrificial pixel nearest my controller.

My Config:
* Total LED Count: 142
* One LED output:
  * Pin: 2
  * Start: 0
  * Count: 142
  * Reverse: ON
* Skip First LED: ON

Segment 0:
* Start LED: 0
* Stop LED: 142
* Reverse direction: OFF

With Skip First LED on and Reverse **off**, my sacrificial pixel is dark and all 141 hanging pixels are lit, and animations like Halloween appear to move from right (controller end) to left (far end)
e.g. When facing the house, Halloween orange `o` and purple `P` segments move from right-to-left, and the sacrificial pixel is dark `x`:
```
Skip First: ON
Reverse: OFF
<far end              sacrificial  controller>
PoooooPPPPPoooooPPPPP      x
oooooPPPPPoooooPPPPPo      x
ooooPPPPPoooooPPPPPoo      x
oooPPPPPoooooPPPPPooo      x
```

I think I want these animations moving from left-to-right (from far end toward the controller end), so I turn on Reverse in LED & Hardware Setup (because this string of lights is physically installed "the wrong way", this feels like hardware setup).

With Skip First LED and Reverse both **on**, Halloween orange `o` and purple `P` segments move from left-to-right, but now the far pixel is dark `x` and the sacrificial pixel is animated:
```
Skip First: ON
Reverse: ON
<far end             sacrificial  controller>
xPPPPPoooooPPPPPoooooP    P
xoPPPPPoooooPPPPPooooo    P
xooPPPPPoooooPPPPPoooo    o
xoooPPPPPoooooPPPPPooo    o
```

**Expected behavior**
I think I would expect "Skip First LED" to always skip the first LED electrically/physically (nearest the controller), and then the reversal would affect the remaining LEDs.

```
Skip First: ON
Reverse: ON
<far end             sacrificial  controller>
PPPPPoooooPPPPPoooooPP    x
oPPPPPoooooPPPPPoooooP    x
ooPPPPPoooooPPPPPooooo    x
oooPPPPPoooooPPPPPoooo    x
```

As stated above, it looks like there are several other ways to accomplish what I want (animations starting at the far end), but in my case I have a sacrificial pixel and a string of lights mounted "backwards", and setting both of those as hardware setup options seems logical.

**WLED version**
 - Board: ESP8266 NodeMCU
 - Version 0.12.0 2104020
 - Format Binary

**Additional context**
Our lights were the best Halloween decoration in the neighborhood, thanks to WLED and DrZZs!

blazoncek, 2021-11-01:
You are using a version with a known bug regarding sacrificial pixel and multiple segments.

Try with ignoring the "reverse" in LED settings (uncheck it) and apply reversing in segment instead. This should fix your issue.
The other option is updating to 0.12.1 or latest beta 0.13.0-b4

I will take look about the behaviour you describe nevertheless.

timkingman, 2021-11-08:
Thanks, @blazoncek . v0.12.1-b1 sound-reactive still has the issue I described above, but v0.13.0-b4 behaves as I would expect, and 0.13's updated LED settings (including per-output Skip First checkbox) are easier to understand.

Feel free to close this issue.

closed 2021-11-08 by blazoncek