wled/WLED #2305: Changing content of Index.htm

makeroutlet, 2021-11-01:
Followed this howto https://github.com/Aircoookie/WLED/wiki/Add-own-functionality

When i run npm install and npm build all changes work except for changes to index.htm. Is this something im doing wrong or does this process not update the contents of index.html anymore?

Aircoookie, 2021-11-01:
Hey!
The index.htm is cached by default to improve performance. Before updating your changes to the ESP, make sure to increment the version code in line 11 of `wled.h` by one (or use Ctrl+F5 to hard-refresh in browser).
I hope that solves the problem :)

makeroutlet, 2021-11-01:
so increment this number? #define VERSION 2109220

makeroutlet, 2021-11-01:
does the increment change allow the index.htm to update through the npm install and npm build commands?

Aircoookie, 2021-11-04:
Yes!

No, the npm build will most likely update the index.htm correctly even without it, but your browser will still have the earlier version cached. Changing `VERSION` will tell your browser that it will need to refresh the index.htm

closed 2022-04-16 by Aircoookie