wled/WLED #2307: IR json hang

hoddl, 2021-11-02:
I have an IR 38key with ir.json implementation.
Every key works fine while the effect is on "Solid". 
FP=0 or something else switch the effect to "Gradient" and the IR Remote dose not work. 
I wodere because FP means palette and not effect.
Same when FX is used the effect changes and Remote do not work.  On the WebUI switch to "Solid" and the IR work again.
Some Keys are the same code as 24key RGB remote, so I tryed  with the precompiled IR Remote Version and the same effect. Remote only works when Effect ist "Solid"

**To Reproduce**
Factoryreset, LED Key =GPIO 5, IR = GPIO 4, ir.json upload from https://kno.wled.ge/interfaces/json-ir/38-key_ir.json

**WLED version**
 - Board: Wemos D1 mini
 - Version 0.13.0-b4, dev200603]
 - Format  Binary

blazoncek, 2021-11-02:
Create your own ir.json file and experiment.

hoddl, 2021-11-02:
Thank you for the answer. When the effect is diffrent than Solid, the IR do nothing, independet of ir.json. It is possible to create an ir.json for all key that work, but when the effect is change the IR do not work so it must be some thing in the IR do not receive some thing when the effect is change.

Aircoookie, 2021-11-02:
Interesting, my theory would be that if an effect is running, the loop is not called often enough and the IR signal is missed (though it should be interrupt based?). Just speculation though.

@hoddl how many LEDs are you running?

hoddl, 2021-11-02:
Im still  testig so only 30 LEDs are runnig. Interrupt problem seems the right way but I have no Idea for a solution.

blazoncek, 2021-11-02:
It does work on all my installs from ESP01, Wemos D1 Mini and ESP32. No issues there.

Aircoookie, 2021-11-02:
I think LED pin 5 might make the difference here. Pin 2 is really the recommended on ESP8266 for hardware driving, 5 uses software bitbang and is thus more (too?) taxing on the CPU.

@blazoncek I don't currently have an IR tester, can you quickly test whether setting LED output to 5 causes this issue?

blazoncek, 2021-11-02:
Very unlikely today, but I have a few installs where IR works on GPIO 3 and GPIO12-14.

Aircoookie, 2021-11-02:
GPIO3 is also hardware, but if 12-14 work (on ESP8266), no reason 5 shouldn't.

blazoncek, 2021-11-02:
@hoddl build with debug flag enabled and attach USB cable and monitor serial output.
When you press button on IR transmitter it should be displayed on serial output.

hoddl, 2021-11-02:
@Aircoookie thank you, I change the LED PIN to GPIO2 an now it seems to work.

closed 2021-11-02 by hoddl

Sojourneer, 2024-03-28:
Switching from GPIO4 to GPIO2 also worked for me, even with ESP's recommended 12K pullup on GPIO2.  The refresh rate is an order of magnitude higher.