wled/WLED #2308: Sync Spacing & Grouping Settings Across Multiple Controllers

mattkearl, 2021-11-02:
I have multiple controllers and use the sync function to pass the current preset over to multiple sets of lights. It would be amazing if we could **sync the spacing and grouping settings over to other controllers**. 

I have some presets that look really good with various spacing and groupings, but because sync does not support that feature, it looks really bad. I understand how and why syncing segments would not work, but spacing/grouping should be able to be sent over if they are separated away from individual segments into a master spacing/grouping arrangement. One solution might be to sync the first or master segment spacing/grouping over to the other controllers. 

Either way, I feel this feature would be really helpful for those that use other patterns/spacing/grouping and need that to be uniform across multiple controllers and light patterns (think of exterior Christmas lights).

blazoncek, 2021-11-02:
Use virtual LEDs available in 0.13 if you have less than 10 controllers.

mattkearl, 2021-11-02:
> Use virtual LEDs available in 0.13 if you have less than 10 controllers.

I am running 0.13.0 b4, and can't seem to find any options labeled virtual LEDs. Were you referring to virtual LEDs in this article (I think you wrote)? https://github.com/Aircoookie/WLED/wiki/Segments

I understand that presets will save virtual LEDs (spacing and grouping), but when syncing, it only sends color and effects to other controllers. Am I missing something here you are referring to when talking about a virtual LED?

Thnx in advance. You do great work.

blazoncek, 2021-11-02:
No.
Segments can be used in combination with virtual output (DDP network, look into LED settings, LED type).
This has nothing to do with Sync but instead relies on one ESP controlling directly each LED on other controllers.
For help, please use official Discord.

closed 2022-01-25 by blazoncek