wled/WLED #2309: Highpower LED Chip 10W 350mA

DarthWeber, 2021-11-02:
Hi, is there any way to drive those high power LEDs which needs 350mA constant current with WLED ? 

**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Aircoookie, 2021-11-02:
Can you send a link to the LED you want to use?
Most likely you can drive it with WLED, but will need extra circuitry.

blazoncek, 2021-11-02:
That is what I use.
You need MOSFET as IRLZ044N.

DarthWeber, 2021-11-02:
@aircookie for example https://www.ebay.de/itm/161057358276?mkevt=1&mkcid=1&mkrid=707-53477-19255-0&campid=5338364437&customid=161057358276_12576&toolid=11000

@blazoncek Is there any circuit diagram ? Or pics ? Not sure which board i should use and how to connect the MOSFET

DarthWeber, 2021-11-03:
@blazoncek Thanks, but i am not experienced enough to get this connected to my board - above that i expect this will be driven by PWM and since i have at least two LEDs, i do not know which board is suitable.

In the meanwhile i found [https://www.led-genial.de/Power-Dot-10W-WS2811-kompatibel](url) which seems to be pretty easy to use with the Wemos mini board.

blazoncek, 2021-11-03:
Look for @srg74 [ESP-01](https://www.tindie.com/stores/serg74/) board on Tindie or @intermittech [QuilnLED-Quad](https://quinled.info/analog-led-dimming/#AnalogBoards) controller. They offer PWM circuitry.

blazoncek, 2021-11-20:
[image]

pyr0ball, 2021-11-24:
try one of these! 
https://www.aliexpress.com/item/32798198341.html

If you need something like the above, but a different form-factor, you'll need someone to design you a custom carrier.

blazoncek, 2021-11-24:
> try one of these! https://www.aliexpress.com/item/32798198341.html
> 
> If you need something like the above, but a different form-factor, you'll need someone to design you a custom carrier.

These are digital. Mine is analog.
I guess OP wanted analog LEDs.

closed 2022-04-16 by Aircoookie