wled/WLED #2310: Upnp woes

feraldruid22, 2021-11-02:
My laptops kernel log flooded with this crap coming from the WLED node:

```
[142933.218747] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.219793] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.223638] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.224122] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.225003] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.225004] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.226923] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.227402] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.232083] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142933.233162] IPv4: martian destination 0.0.0.0 from 192.168.2.2, dev wlan0
[142963.322375] net_ratelimit: 15 callbacks suppressed

```
```

tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
23:17:13.356389 IP 192.168.2.2.1900 > 0.0.0.0.0: UDP, length 0
23:17:14.354623 IP 192.168.2.2.1900 > 0.0.0.0.0: UDP, length 0
23:17:15.349117 IP 192.168.2.2.1900 > 0.0.0.0.0: UDP, length 0
23:17:16.352105 IP 192.168.2.2.1900 > 0.0.0.0.0: UDP, length 0

```

WLED version 0.13.0-b4

Is there a way to turn this off I didnt see any option for it, I will have to put the device into another wifi network if this doesnt stop very annoying.

huggy-d1, 2021-11-02:
Do you have MQTT enabled / configured in WLED?
More information on your WLED configuration might get some answers.
Was a browser up with a tab pointing to WLED?

Aircoookie, 2021-11-03:
Seems like it could be a reply to an SSDP M-SEARCH request sent by another device on your network. Is the Alexa support in WLED Sync settings enabled? If yes, disabling it and rebooting will likely stop these packets.

feraldruid22, 2021-11-04:
Thank it was the ALEXA after disabling that and rebooting I no longer receive these packets.

closed 2021-11-04 by feraldruid22