wled/WLED #2313: Enhance AP mode only

KlausHans, 2021-11-03:
**Is your feature request related to a problem? Please describe.**
Hello, i am using Wled on my RC Car. I get very cool effects with Wled. Thank you for that fine piece of software.
However, since i have to use Wled in the access point mode only, i stumbled upon some barriers in terms of ease of use:
- the welcoming page, where i have to choose to go to wifi settings or to the control ui is the main screen and i get always here if i change a setting and use "back" or something similar. The main screen should be the controls.
- the captive portal function wled uses in ap mode is annoying. I always have to tell android, that i want to use the connection as is. If i don't do that i can't open Wled in the browser.

**Describe the solution you'd like**
I would like some sort of explicit ap only mode or more options to tick to get there. The captive portal function should be deactivatable and the controls should be the main screen, not the welcoming page. 

**Describe alternatives you've considered**
The alternative is to live with the current state which means quite a bit of overhead if i connect to Wled.

**Additional context**
[image]

That would be great, thanks!

Aircoookie, 2021-11-04:
Hi!

To disable the Welcome page, simply delete the default network name `Your_Network` in WiFi settings. Then, the controls will be the default page!

For the second issue, I could consider adding an option to disable the captive portal, and/or change the IP used in AP mode.

mjg1088, 2021-11-04:
For my portable project, I have used a second 8266 as an access point. I connected my wled project to that access point and I am able to use the app on either my phone or PC pretty seamlessly. Maybe consider adding another node to get out of the captive portal!

KlausHans, 2021-11-04:
Deleting the default wifi string worked indeed (after reset), thank you! Maybe you could add a little hint somewhere? Or did i missed one?

That option regarding the captive portal would be great.

Adding another node makes no sense in my case (rc car). 

I am planning to make kids lantern with wled, another use case for a good ap mode only. :)

keukenrol, 2024-12-23:
Are there any updates on this? I am also using my android phone with a standalone wled setup. Being able to jump onto the default controls page in ap mode would be nice.

louisnorwood, 2025-01-31:
This trick no longer works. When deleting the Your_network WiFi name, after reboot it comes back.
Is there some validation in the latest version that is causing this?

DedeHai, 2025-01-31:
this is the code:
```
if (strcmp(multiWiFi[0].clientSSID, DEFAULT_CLIENT_SSID) == 0)
    showWelcomePage = true;
```
so should still work

DedeHai, 2025-08-17:
just rename it to whatever you like should work as the code i posted indicates.