wled/WLED #2314: Full Reset

Imperattor, 2021-11-04:
How to do a full reset, if I changed and forgot the password in the settings and now haven't access to the control interface ?
Help, please!

blazoncek, 2021-11-04:
Attach ESP to computer using USB cable and erase flash then re-flash WLED.

Imperattor, 2021-11-04:
> Attach ESP to computer using USB cable and erase flash then re-flash WLED.

i know. impossible now. 
and other methode?

blazoncek, 2021-11-04:
remember password

poelzi, 2021-11-16:
Pressing user button 6 seconds opens AP with default password wled1234 or something like that.

Imperattor, 2021-11-16:
> Pressing user button 6 seconds opens AP with default password wled1234 or something like that.

what the button ??

poelzi, 2021-11-16:
There are 2 buttons on most esp boards, one reset one USR1 button. Press the USR1 button 6 seconds and look for the wled wlan. if not, your firmware was to old for that feature.

Imperattor, 2021-11-16:
> There are 2 buttons on most esp boards, one reset one USR1 button. Press the USR1 button 6 seconds and look for the wled wlan. if not, your firmware was to old for that feature.

ESP-01 have a any button?

blazoncek, 2021-11-16:
> ESP-01 have a any button?

I feel your pain but actually they do, if you invested in them.
You will not find a better project: [ESP01 shield](https://www.tindie.com/products/serg74/esp-01-wled-shield/)

Imperattor, 2021-11-16:
> > ESP-01 have a any button?
> 
> I feel your pain but actually they do, if you invested in them. You will not find a better project: [ESP01 shield](https://www.tindie.com/products/serg74/esp-01-wled-shield/)

of course, but i can't physical connect to the board (((

blazoncek, 2021-11-16:
I bet you can. Not without hassle as I can see. 😉

closed 2022-02-04 by blazoncek