wled/WLED #2315: Enabling analog buttons breaks WiFi on 0.13.0-b4

ajh51, 2021-11-04:
### What happened?

Running beta version 0.13.0-b4 on D1 Mini

Enabling analog "button" with potentiometer wiper connected to pin (A)0 kills the wifi connection to the part.

The analog function works fine - tried global brightness and colour hue (setting 200) - both work great 

However the standard WiFI connection stops working.  WiFi Router shows that the WLED instance has gone from the network.  Need to re-install to recover.

Tried with the AP also - same effect.  AP connection works until the part is configured but as soon as the analog button is selected the WiFI drops - again re-install needed o recover

Tried with no potentiometer connected - to rule out wiring issue (although potentiometer function works so wasn't expecting much) - no change

Reset/power cycling of the D1 Mini board doesn't help.

### To Reproduce Bug

Clean install of 0.13.0-b4

set button 0 to input 0 and to mode "analog"

AP/Wifi drops within a few seconds and cannot be recovered

### Expected Behavior

WiFi is unaffected by selecting analog button

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b4 build 2110110

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

Thanks for any help!!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-11-06:
None of this is happening on my test units.
What is the resistance of your potentiometer.

I would also suggest to add a resistor of 10% value of potentiometer in series with it connected to 3.3V (so: 3.3V -> R1 -> potentiometer -> GND).

ajh51, 2021-11-07:
Thanks for trying it out for me.

Hmmm....   I'm sure the pot isn't the issue, I'm using a 25k lin pot.  No series resistor but the issue happened at mid point as well as at extremes so I'm sure that's not it.

I spent some more time debugging - I stripped the system down to just WLED and the potentiomenter and the issue does go away.  Adding things back in it is the remote control that I have connected that breaks it. 

I'm implementing a wired remote using the IR port driven by an Arduino aping the (unmodulated) NEC protocol that the IR port expects to see.  It works great and all is well until the potentiometer is connected. 

I will dig out a TSOP38238 and try that to put WLED in the clear then double check my logic levels again.

Can we keep this open until I tried those things?

ajh51, 2021-11-11:
I've tried this out with an actual open collector with pull up driving the IR remote pin and I cannot reliably get the potentiometer and IR remote to work together.  Both work fine independently but together they sporadically break the wifi connection.  Sometimes this is temporary and sometime permanent requiring reinstallation.  
[image]

blazoncek, 2021-11-12:
If you really want some help, post your setup, please.
Dump cfg.json from /edit page or from backup in Security & Updates

lununez, 2022-02-01:
@ajh51 Hey there - did you ever figure this out? I’m experiencing the exact same issue. Wi-Fi dies and is unrecoverable without a feel reinstall once I set up the potentiometer (but even without setting up anything else)

Aircoookie, 2022-02-01:
Is this issue still present on b6 or master?

rivelta, 2022-05-08:
[image]

how i can add more than 2 analog button?

blazoncek, 2022-05-08:
Use ESP32.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

softhack007, 2023-01-12:
Loses due to inactivity.

closed 2023-01-12 by softhack007