wled/WLED #2316: Slow boot and afterwards turning on, off, on

Cusify, 2021-11-05:
Hi!
I have been using WLED to control my light since 2019. When I press the light switch, then the light turns on. Therefore, it is current dependent.

Since version 0.13.0-b0, and most noticeable in 0.13.0-b4, it takes an eternity (about 2-3 seconds) until the light turns on. If it is then on, then it goes off again and then on again.

I have now turned off the crossfade in the settings of the LEDs. Thus, I have at least no more flickering.Nevertheless it takes a long time until my ESP8266 boots (recognizable by the blue LED when it lights up). 

Does anyone have an idea what this can be? If I run the ESP on the USB, then he starts faster :/

Translated with www.DeepL.com/Translator (free version)

blazoncek, 2021-11-06:
Eternity is much, much longer than 2 or 3 seconds. 😉

Most likely your PSU (either external or the one on ESP, converting from 5V to 3.3V) is inadequate. And causes brownouts during boot.

sneak-thief, 2022-02-12:
I just tested and am getting the same slow turn-on time with an ESP32. 

I'm using SK6812 strips that range from 85 - 255 leds. The power supplies can deliver more than twice as much current as the strips can draw at maximum brightness. 

2-3 seconds is an awfully long time to wait for a light to turn on after turning the switch on.

blazoncek, 2022-02-12:
> Something clearly changed after b3. Was it maybe the strip init code? [8b2145b](https://github.com/Aircoookie/WLED/commit/8b2145bd888ee930fcb9ad76139d39812586ae3a)

A lot has changed since b3. 😁 There is more functionality and flexibility than ever and this has contributed to boot time. It will be difficult to make it any shorter.

If you want, you can check the change log and try to pinpoint the change that extended the boot time considerably (use git to checkout certain commits and build the code manually).

sneak-thief, 2022-02-13:
The changes occurred between b3 and b4. I'll checkout the individual commits and verify which one caused the slow boot.

The other thing I noticed is that with b3 and prior, when you turn on and boot my strip fades in as per the transition settings.

However after b3, transitions on power-on don't work.

Even if you set a long transition and turn on power to the mcu & led's, it's still off for almost 3 seconds then strip goes to full brightness of the preset.

softhack007, 2023-10-07:
Closing, as this topic was inactive for a lo g time. The parts of WLED related to described problem were largely rewritten in the meantime. Please open a new bug report if you still experience the same problem with 0.14.0.

closed 2023-10-07 by softhack007