wled/WLED #2317: HI

Adm1rall, 2021-11-06:
Good time =))
I really like wled

1 to save electricity, add the function of disabling wifi or disconnecting after a specified time
2 remember the last mode turned on when the power is completely turned off

Aircoookie, 2021-11-06:
Hey!
1) Then how would you turn it back on? If you don't need it for a long time, I would recommend turning off the power to WLED entirely by unplugging it or using a timer or smart plug (although they also use a little bit of power). The standby draw of the LEDs is likely higher than that of the ESP, so you can reduce standby power draw by adding a relay to switch off the power to the LED strip: https://kno.wled.ge/features/relay-control/

 2) This is possible with the auto-save user module: https://github.com/Aircoookie/WLED/tree/master/usermods/usermod_v2_auto_save (requires a custom compilation, see https://kno.wled.ge/basics/compiling-wled/)

Adm1rall, 2021-11-06:
thank you for responding to me so quickly)

my device is powered by 18650
this is a dog collar
I need to disable wifi to save energy and so that wifi does not harm the dog

Adm1rall, 2021-11-06:
https://www.instagram.com/p/CJjYBdyszN5/
это ошейник на ардуино хочу переделать его под esp 8266 с возможностью отключения wifi например через 5 минут после включения

Adm1rall, 2021-11-06:
> 
> 
> https://www.instagram.com/p/CJjYBdyszN5/ this is an arduino collar I want to redo it for esp 8266 with the ability to disable wifi for example 5 minutes after switching on

Aircoookie, 2021-11-06:
Nice project and cute dog!
When on displaying effects, the LEDs will use WAY more power (say 1A) than the ESP8266, which should enter wifi standby mode if not receiving commands (20mA). When not using the collar, i would strongly suggest either disconnecting the battery or adding a hardware switch to cut power to both the LEDs and the ESP. Otherwise, your battery will be deep-discharged within a few days (not because of WiFi, but because of the LED standby power draw)

WiFi won't harm your dog, same as your phone does not harm you.

You can fully disable wifi by not setting a network name and setting `AP opens` to "Never" in Wifi settings, however you will need to add a pushbutton between gpio0 and ground that you can press for 7 seconds to turn on the WLED-AP again.

I might consider adding an option to turn off the AP after some minutes if no device is connected to it, but as mentioned before it is not dangerous and does not use a lot of power.

Adm1rall, 2021-11-06:
Thank you =)) in winter it is dark here, you need to see the dog from a long distance, this is her safety

I don't want to take the risk and check on the dog vredel and wifi =)
when the collar is not in use + the battery will be disconnected

I want wifi to work for a couple of minutes when the power is turned on, so that I can configure the glow mode and then after 5 minutes it turns off or turn wifi on and off, for example, from a physical button, and it is desirable that wled remembers the last mode turned on

Adm1rall, 2021-11-14:
Good afternoon, when can we expect a new version?

allenhuffman, 2022-12-18:
We are looking for #1 as well. The idea is to have an access point available for a set time on power up, then it shuts down until the next reboot. This means in normal operation, there is no Internet happening, but a power cycle allows connecting for configuration changes.

If there are any API calls to enable/disable the WiFi, that would suffice since they could be done in a usermod.

blazoncek, 2022-12-19:
I created a `power-ap` branch with powerap usermod which allows you to disable AP altogether and then power-cycle ESP a few times to enable AP mode.
It may get incorporated into WLED if there is enough interest.

allenhuffman, 2022-12-19:
> I created a `power-ap` branch with powerap usermod which allows you to disable AP altogether and then power-cycle ESP a few times to enable AP mode.
> It may get incorporated into WLED if there is enough interest.

Were there changes to things beyond that powerap.h required to make that method work? Is there just a function that will disable WiFi? Ideally, one could have it in normal mode (if WiFi defined, try to connect; if it can't, if AP defined, turn on AP) and just turn it off after time. If on, and there is a connection, it could disable the shutoff for the remainder of that power cycle.

allenhuffman, 2023-01-13:
> I created a `power-ap` branch with powerap usermod which allows you to disable AP altogether and then power-cycle ESP a few times to enable AP mode.
> It may get incorporated into WLED if there is enough interest.

How do we need to set this up to make it work? Specifically, I want to use your lines:

`apBehavior = AP_BEHAVIOR_ALWAYS;
WLED::instance().initAP(true);`

...in my usermod setup().  Ideally, unit has WiFi disabled, but if an I/O is held down during power up, I want to have it turn on.

jmerc77, 2023-07-12:
I have a thought. why not disable the AP as long as it is connected to a wifi router. but when it looses signal it waits a little and then enables it back. the idea is mainly for security as i am discovering that there seems to be no ssl possible with softAP. though can not confirmed or deny as there is literally no info about it that i could find.

blazoncek, 2023-07-12:
@jmerc77 that's already available.

barrymossel, 2024-01-06:
> I created a `power-ap` branch with powerap usermod which allows you to disable AP altogether and then power-cycle ESP a few times to enable AP mode. It may get incorporated into WLED if there is enough interest.

Was this ever incoporated?

blazoncek, 2024-01-06:
The feature (1) requested is available in the `temporary-AP` branch which may be merged if there is enough feedback.
This is alternative to `power-ap` which works in a different way.

robhaswell, 2024-01-08:
Hi, we have a use-case for this as well. We are using WLED to power runs of lighting for our night-time drone racing events. Racing drone control links are susceptible to 2.4GHz interference which means we must disable the APs during the event. However, it would be convenient to be able to access the AP for settings changes.

I found a thread that said that pulling one of the GPIOs low would enable the AP if it's disabled, but we have to make 14 control units and adding a button would greatly increase the complexity. Having the AP available for e.g. 2 minutes after power up would be perfect. Happy to hear other suggestions though.

robhaswell, 2024-01-25:
@blazoncek I'm using `temporary-AP` for our project with great success. I made a couple of changes:

1. I set the timeout to 1 minute as it was more suitable for our purposes.
2. I made `AP_BEHAVIOR_BOOT_NO_CONN_5MIN` the default behaviour.

I made change (2) initially because the option was not appearing for me in the AP configuration UI, despite me being able to see it in the source. I don't know why that is. Either way, I am flashing dozens of devices so setting this as the default was desirable. It would be nice to be able to have that ability via platformio_override.ini.

Thank you for this branch.

blazoncek, 2024-03-24:
implemented in #3721 and available since 0.15.0-b1

closed 2024-03-24 by blazoncek