wled/WLED #2318: Inplausible IR values after activating an effect

DrunkenHeroGH, 2021-11-06:
### What happened?

After activating an effect with the IR Remote Control, the IR Values are inplausible.

### To Reproduce Bug

Little example:

1. Press "red" on the remote control (OK)
 -> Serial output: IR recv;0xF720DF

2. Press "green" on the remote control (OK)
 -> Serial output: IR recv;0xF7A05F

3. Press "flash" on the remote control (OK)
 -> Serial output: IR recv;0xF7D02F
 -> correct mode is activated (Colortwinkles)

4. Press "flash" on the remote control again  (not OK)
 -> Serial output: IR recv;0xE0F631A6

5. Press "flash" on the remote control again(not OK)
 -> Serial output: IR recv;0x2D98EA73

6. after changing the effect to "solid" over the web interface, the IR code is received correct again.(OK)

hardware:
1. white 24-key IR remote with R,G,B + 12 color-tones
2. D1 mini
3. TSOP 31238 IR-Empfänger-Module

### Expected Behavior

The correct IR Values are always received.

### Install Method

Self-Compiled

### What version of WLED?

0.12.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2021-11-06:
Thank you for the report!
What is your LED pin(s)? Sounds like it might be the same issue as #2307

DrunkenHeroGH, 2021-11-08:
Thanks for the hint. After connecting the LED Pin to GPIO2, everything works fine.

closed 2021-11-08 by blazoncek