wled/WLED #2325: E1.31 (DMX) group pixels

KevinVanthuyne, 2021-11-08:
I am in the process of making a stage lighting installation with pixel tape and WLED. So far it's working really well, but controlling all pixels (60 leds/m) is eating up a lot of channels and universes. I don't need to control each led separately, in my case about 10-20 "groups" of 3 or 6 leds per meter would be enough. This would drastically reduce the amount of pixels to output and thus make it possible to run faster animations and maybe a cheaper license for the outputting software. Is this something that is possible?

Imperattor, 2021-12-04:
Hi! tell me please, how did you control the lamps via wifi and what program with what settings? I just can not understand how to control the lamp by DMX ?

max9901, 2021-12-06:
Hi kevinVanthuyne, i created a small exploration edit on my fork that has an extra dmxMODE which can controll all segements seperatly via dmx. Which will give 14 channels for each segment. I u like u can try it out. I'm planning on building this out to a more robust version later. Would u agreed with this approach to be nice for stage lightning ?

https://github.com/max9901/WLED/tree/feature/multi-dmx

Let me know if u want to excange tought about this subject. !

deggle, 2022-05-21:
Hi - I think this is essentially very similar to what I'd asked for here: https://github.com/Aircoookie/WLED/issues/1663

I actually came on today to see about re-opening that ticket, but did a search first and found yours...

I think the changes I'm suggesting there would work well for us both? The only other thing I did think about was adding an additional first channel as an overall master dimmer, then follow that with all the segments RGB(W) channels. I think this would be beneficial if you wanted to use a fader on a console to dim a whole fixture.

I also can see the benefit in also adding a third new option - "Segment Effect" - which I assume would work like the branch from @max9901 allowing much wider control. While nice, I do think just the RGB(W) per segment would be a really great start and cover what many of us may want for external or lighting console control.

I have had a look at the code around this (as per my linked ticket above) and I am thinking this would be a relatively straightforward code change/addition for someone who knew the workings of the sACN/DMX functions (I did take a look but didn't want to bodge it).

It's also probably not something that will add a lot of code or bloat the project as the inerrant code is all there already, it's just about doing some loops for the segments. I do think it would be really useful for those of us that was to use WLED for some cool stage/architectural/etc type fixtures. 

A good simple example would be something like a 4 sided square, or isometric cube shape design. Each side/line would be set as a segment then easily controlled - however trying to expose all the individual LEDs would need 100s of channels which makes controlling them via a lighting console or similar quite a challenge (or almost impossible).

@Aircoookie I'm not sure how much time would be involved in this - but I'm happy to put up £100/€100 if it would help get this into WLED as it would be very useful for me and also potentially the wider community.

Ps. Another potential thing that would cover some of these use cases, would be preset recall via DMX as that could pull in segments with fixed/pre-defined colours (and effects). It could also use less channels per device - perhaps just 4-6 based on (D)RGB(W)P where on value change P would recall a given preset. To keep this simple, I think the RGB values would just have to be limited to the first segment if there were multiple defined. How about another £100/€100 if we can get that in too.

SkylarTheGrey, 2022-06-09:
> Ps. Another potential thing that would cover some of these use cases, would be preset recall via DMX as that could pull in segments with fixed/pre-defined colours (and effects). It could also use [...]

@deggle I would also love this, I laid some of it out in #2556 that is about UDP preset sync triggering and would also chip in to fast-track it. Our concept would use console dry contacts to send UDP sync packets and also get the power of sync groups.
Presets trigger over DMX would be another way to solve a bunch of issues we have currently with E.131 effect mode not being able to use the full power of segments. WLED in the WUI is incredibly powerful and we would love to trigger that from our lighting console.

We even have concepts we cannot yet implement where we have devices that have DMX preset trigger and local button preset trigger intermixed for props and set pieces.

blazoncek, 2022-06-09:
Grouping pixels with live (aka realtime) data is available in recent master by using _mainsgment only_ setting in Sync settings.
It was introduced in 6079effae3dcb86230d235a2759b4813026b0855

closed 2022-06-09 by blazoncek

deggle, 2022-11-08:
Hello - sorry, I've only just seen this reply. I will be checking this out when I get chance in the next few days. Is it on the latest release binaries?

Aircoookie, 2022-11-08:
Yes, it is part of 0.13.3 already.

lorandnagy, 2022-11-08:
Hi there! I have 50 Leds, 4 segments. Artnet configured as multi rgb. If I check main segment only, just segment 0 is receiving realtime data, but as multi rgb.
I tought this topic is about to control one segment as one Led (3 channel for one segment). This way we could control 4 segments for example with only 12 channels. 
My main problem is:
I am using rekordbox with RB-DMX1 device. This is able to send out just one universe with 512 channels.
I would be happy to work on this, just was currious if there are some steps in this dirrection.

deggle, 2022-11-08:
Unfortunately this is not achieving what I think most of us were looking for.

This feature/checkbox allows the DMX data to be limited to just the first Segment (0) and leave the others independent. This may be useful for some, but I can't really think of a scenario where it's that useful and I can't see if/where it's been requested.

What I think people are looking for is rather than the DMX channels being something like:

1. Master Dimmer (optional channel)
2. LED 0 R
3. LED 0 G
4. LED 0 B
5. LED 1 R
6. LED 1 G
7. LED 1 B ... and so on

It would be:

1. Master Dimmer (optional channel)
2. Segment 0 R
3. Segment 0 G
4. Segment  0 B
5. Segment 1 R
6. Segment 1 G
7. Segment 1 B ... and so on

Here's some other requests asking for the same https://github.com/Aircoookie/WLED/issues/1254, https://github.com/Aircoookie/WLED/issues/1115

I would see this being an additional option (or two) on the "DMX Mode:" dropdown, perhaps titled "**Segment RGB**" and perhaps "**Dimmer + Segment RGB**" (perhaps with RGB+W variants). Maybe even a "**Segment Dimmer + RGB**" which would be 4 channels (5 for W) and include a dimmer channel for each segment (may be useful for lighting desks).

Example usage scenario:

Imagine you have a square with 50cm sides around which 2m of LED strip is run with 60 leds per meter. That's 120 leds so would need 120 x 3 = 360 (plus optional dimmer channel, and with RGBW you're now over 480  channels). However if the design/effect you want only needs to control the colour of each side of the square, this could be segmented and driven with 4 x 3 = 12 channels - many many less (even if we added an additional dimmer channel for each segment it would only be 16).

I'm using 144/m LED tape to reduce the 'dot effect' and with RGBW an entire universe is used up in 88cm of tape so it's just not workable. Being able to divide it into fractions based on how it's being used would be so helful.

Some time ago I did start a thread with my initial code tweaks/thoughts, but gave up at the hard bit!

https://github.com/Aircoookie/WLED/issues/1663

The £100 bounty remains!

max9901, 2022-11-08:
hi I am still very interested in the feature,

the first crude coding I did on the fork above was under quite a bit of time pressure to finish a light art installation, which btw worked perfectly for a week straight with this feature, the only problem I have with the solutions is that is is really hard to manage the segments correctly, especially as one can overlap segments and have active and inactive segments, (as far as I remember from coding this). I will try my best to look into this again when I find the time but no promises what so ever here.

If somebody can figure out or explain me how segment management works in the code that would be a massive headstart to get this (these) features working.

max9901, 2022-11-08:
hi @deggle, I create a very simple crude test for u to try u can get the code here:
https://github.com/max9901/WLED

do not change the DMX mode ! as I did not update the web interface fully yet I think.

the current code does:
dmx channel:
1: Segment 0 R
2: Segment 0 G
3: Segment 0 B
4: Segment 1 R
5: Segment 1 G
6: Segment 1 B ... and so on

no master dimmer! but we can include that later if u want the same for a white channel!

**PLEASE NOTE THAT U ARE RESPONSIBLE TO NOT HAVE OVERLAPPING SEGMENTS** otherwise it works a bit weird. 

If needed I can be of assistant to get this on your setup, I tested it on a led strip at home and it works for me. If this is the solution u are searching for I can finish up this code properly and make a PR. 

cheerz,
Max

deggle, 2022-11-09:
Hi, taking a look now - I think it's building...
(I'm a pretty competent .net/VS dev, but a but of a noob with VS code/c stuff)
Will let you know how I get on...
Tim

lorandnagy, 2022-11-09:
@deggle Hi! I wish you luck! How can I contribute to achieve the goal? I am newbie with github and not really pro in VS, but if i can iwould love to help

deggle, 2022-11-09:
Hi @max9901,

So I managed to get this to build and deploy. The device (ESP32) certainly got the update as I've not seen some of the newer features before (noted the 2D stuff in that build - need to check that out). I created 3 segments and sent some DMX (ArtNet and sACN tried) data but it didn't really work and after some time just set the whole lot of LEDs to the RGB value.

It occurred to me that all the settings from my prior WLED version were maintained (different section/file system I assume) so that may have made the config deviate from what you're using to dev/test so I did a factory reset. However, it no longer showing as an AP so I've not been able to connect to the GUI since.

I've put the latest beta version on and that's brought it back. Actually, I'll try your build again now...

Ok, connected to WiFi with the stock build, now deploying your build.
Configuring the LEDs (64 x WS2812).
Created 3 segments.
With ArtNet it just worked as normal (1 LED per set of channels).
With sACN it did nothing, though I cant be sure if thats an issue on the sending end.

Any thoughts?

Tim

max9901, 2022-11-10:
hi Tim,
Unfortunately, I am traveling so I do not have a lot of debugging options
anymore until after the weekend.

in the Sync settings DMX mode should have an extra option named "Segements
RGB" this is the mode u should have selected, if this is not one of the
options after resetting your device this option should be blank! as I did
not figure out how to fully update the website yet. U can confirm your DMX
mode by looking at the serial monitor from platform IO, just use the upload
and monitor option and this will automatically pop up. After sending a DMX
packet to your device u should receive this message:
"
DMX_MODE_SEGMENTS_RGB
<num-segments>
"

please check if the num-segments corresponds to the number of channels you
have in your instance. also if this message does not show up please fully
clear your device and reflash.

After the weekend if u like i am open for a call to help u get started!

cheerz,
Max

deggle, 2022-11-10:
Hi Max,

I'm confused now as my mode dropdown doesn't show this option...

[image]

Perhaps I built it wrong!?

The only change I made was setting ```default_envs = esp32dev```

Strange!

T

lorandnagy, 2022-11-10:
Hi there! I could also build it. I also configured env:esp32dev, and also have no segment in the dropdown menu.
[image]
Maybe not all devices have this option? I cloned max9901/Wled.

softhack007, 2022-11-10:
General hint on UI files:
if you change files in `wled00/data/`, you have to rebuild the compresses HTML header files otherwise changes have no effect. 

* make sure you have node.js installed on the computer
* open a command window from VSCode, and run `npm run build`. 
* If this is the first time and you see errors messages, you need to do `npm install` before `npm run build`.

https://kno.wled.ge/advanced/custom-features/#changing-web-ui

lorandnagy, 2022-11-10:
@softhack007 thanks!

lorandnagy, 2022-11-11:
I think these are great features, all of the 3. Segment Effects I do not see how I personally could use, because my artnet data is comming out of rekordbox. I was thinking about "cloning" a fixture that is supported by rekordbox, for example a moving head. But as I saw until now there is not a lot of data with what rekordbox could control effects. I will try now to implement my WLED partylights with Segment RGB first. What other good dmx source do you use? I am using an RB-DMX1. Many thanks again!

deggle, 2022-11-11:
Sorry, I think I did not rebuild the HTML hence no option to select. Will try again soon. Yes I think the options you've suggested @max9901 make sense, plus I need RGBW variant as I use a lot of SK6812 strips. I can see some uses for the 'effects per segment' option but I'd only do this if it's not too much work.

deggle, 2022-11-11:
@max9901  Managed to build properly at last (on Windows, needed Node.js, npm install etc) and yes, it works as hoped. I had a huge delay with the change of ArtNet data and the LEDs responding, but I think that may just be because it's on WiFi etc or dev build. The same happened on the normal DMX Mode so I guess it's not a specific issue with the code.

@lorandnagy I'm not familiar with rekordbox but there should be a generic RGB head you can use for the basic segment mode. The WLED effect modes are quote specific and won't be supported out the box, but there may be some sort of 'heard editor' that lets you configure how many channels a fixture uses and what they do. The better lighting desks do this but given rekordbox is more from the DJ angle it may not (but it may well do the sound to light better than a dedicated lighting desk).

Looking forward to the next iteration - would it be possible to get this into the main WLED master branch?

FYI, I just saw this in the pull requests which seems similar but abandoned? https://github.com/Aircoookie/WLED/pull/2199

lorandnagy, 2022-11-12:
Hi There! I made some tests. Yesterday i had also a lot of latency. Today I made a separated AP. No noticable latency!
Thanks a lot!
Videos:
https://www.youtube.com/watch?v=ZiyCV22vJys
https://www.youtube.com/shorts/Y0gnWXnebdg

@deggle yes, in Rekordbox i can manage the channels with fixtures. I was thinking in using something like a "gobo light", but I did not find any automated data for changing the gobo. That would have been great for the effects.

max9901, 2022-11-12:
hello,
@lorandnagy  this is amazing I did not test it yet, but your test setup seems a lot of fun to play with :) Basically, I wrote custom DMX software to do time-coded shows. and paired that with my own (DIY) dmx-nodes (Ethernet->dmx, hardware).  I see find the effects option quite nice to make shows more dynamic but not increasing the channel count to absurd values that one would need for single pixel control... but yeah for record box u should then make recordbox accept WLED fixture.

@deggle  Next itteration will be done somewhere soon next week! afterwards i will open a PR to get it merged into the master. Than it will be up to aircookie and team to decide if they find this valuable enough to actually put into wled. 

cheerz,
Max

mxklb, 2022-11-13:
@max9901 I already implemented the `Segments Effects`, check out https://github.com/Aircoookie/WLED/pull/2881 for testing.

It's still unfinished, needs further testing and maybe so refactoring to be a stand alone DMX mode .. TBD

lorandnagy, 2022-11-15:
@max9901 I will try to find a sollution for effects from rekordbox, but i have to check manually wich fixture gets what kind of data. 

@mxklb I will test it, next time i have time

max9901, 2022-11-15:
@mxklb how did u solve overlapping segments ?

max9901, 2022-11-15:
@deggle I created the WHITE options too, everything is now implemented except for the segment_effect mode. 
https://github.com/max9901/WLED

We also still need to find an elegant way to deal with overlapping segments, just freezing them when they are black is not a real option IMO. but adding an other channel for each segment is also not really an option.

@deggle @lorandnagy  please test the latest and see if u like it.

cheerz 
Max

mxklb, 2022-11-15:
It's not finished yet, but it shall work for one universe and segments count <= 39 (=39*13=507 DMX addresses).

Check diff from wled/main with my fork/branch https://github.com/mxklb/WLED/tree/fxsegs.
Before making PR: I need to make more tests and add some UI things to configure & enable this.
Actually it's implemented as an option/mode of DMX_MODE_EFFECT, search for:

```
  bool useSegementAddresses  = true;  // ToDo: Get from UI -> enable/disable
  uint8_t dmxSegmentSpacing  = 0;     // ToDo: Get from UI -> DMXAddress segment spacing
```
to see actual implementation details.

max9901, 2022-11-15:
@mxklb cool but I guess we did the same I did it differently to minimize the code change needed in the master checkout:
https://github.com/max9901/WLED/tree/dmx-work

mxklb, 2022-11-16:
> 

I use opacity, so I think I don't need to deal with overlapping segments explicitly. This is IMO also more DMX fixture like, than providing an overall brightness channel at DMX start address. Also most DMX gear will do grouping of fixtures including group brightness etc.. Doing so every segment brightness can be controlled using first segment DMX channel. So the only difference of DMX_MODE_EFFECT_SEGMENTS compared to DMX_MODE_EFFECT is, that there are 13 channels per segment and the first segment channel is to be called `Segment Dimmer` instead of `Master Dimmer`.

I finally like to add the `dmxSegmentSpacing` to the UI (when using 'Effect Segment' DMX mode). This is actually still missing and is IMO also a very good feature to handle DMX interfaces with fixed amount of sliders more efficient.

mxklb, 2022-11-17:
By the way, really nice what's going on here. I use WLED with Effect mode for DMX light shows over 1 year now. And I always wanted to be able to control segment effects via DMX. In the past I mainly use DMX_MODE_EFFECT, with whole LED strip. That is why I implemented this DMX_MODE_EFFECT_SEGMENTS. Right after starting implementation, I found this thread, looks implemented slightly similar ..

You should think about avoiding `seg.isActive()` check. Because DMX is a remote control of WLED I don't think users want to use the UI additional to their DMX console. DMX already provides to ability to switch off segments (brightness=0,RGB=0). I think UI options should not be considered when implementing DMX remote control. IMO these checkboxes are only there to control segments via UI. I think everything shall always be controlled by DMX, without the need of the UI once configured. I think it's a really rare use case to switch off/on segments during runtime (and thereby changing DMX segment addresses). But, I don't know .. TBD

And I really think we should combine our changes to improve WLED DMX control. I'll test your changed segment modes when I find time to! I think we should merge it together and also add the `dmxSegmentSpacing` to your new segment modes RGB/W DRGB/W as well. I plan to merge your changes into mine and create a PR .. Alternate I create a PR & you create a separate RP, what do you think?

Finally I finished and tested every thing. It works like a charm :smile:

If someone likes to check it out, `npm install && npm build run` - afterwards build binary for your MC, test.

deggle, 2022-11-17:
I'm not 100% sure I'm following (I've not checked the latest build) but I think I agree with @mxklb in that the UI switching on/off of the segments is probably not relevant and can be ignored from the DMX perspective. Having said that, is there any harm in including it? Certainly if a segment is turned off it should just force black out and not completely ignore it from the addressing structure.

For any use case that would want addresses changing I think you'd just have a pre-set with different segments defined.

It looks like we've got a good bit of added functionality here without an onerous amount of added code - so hopefully when we've completed and tested this @Aircoookie is open to getting it into the main builds...!?

dzelionis, 2022-11-17:
> hi @deggle, I create a very simple crude test for u to try u can get the code here: https://github.com/max9901/WLED
>
> do not change the DMX mode ! as I did not update the web interface fully yet I [...]

Hey @mxklb, i have managed to compile your code with some changes as it was giving me some trouble......but looks like it's not what i am looking for, in my case i looking to have multiple segments, each segment should be controlled as a separate 11 ch DMX fixture (should have its own effect settings etc....) as i am using soundswitch, and that would be best approach as i could control effects via attributes in soundswitch.. and your implementation i think is bit different....isn't ??
I was compiling it on linux, so to get it compiled with no errors i had to modify platformio.ini. That's the file i am using currently https://drive.google.com/file/d/19qirXADyBY03vBtPyV3pZYBJ59yoYWRz/view?usp=sharing 

@Aircoookie, by the way, wled is not discovered via artnet in soundswitch anymore, it was working for a while, not sure if sound switch did some changes or you guys, but something got broken....so i started to use my python script for advertising it again, and that worked out of the box with soundswitch... (https://github.com/dzelionis/artnet-advertiste-wled)

dzelionis, 2022-11-17:
Hey @mxklb, what i see while running your code....i love it already!!!! I think It's exactly what i was looking for!
Thanks

dzelionis, 2022-11-17:
> Hey @mxklb, what i see while running your code....i love it already!!!! I think It's exactly what i was looking for! Thanks

After I have played a bit more with it, i noticed that its not using existing segment sizes, but looks like it is creating new segments..... In my case segments has variable-length as i i have it around the living room (the room is not square shape),  
that's my segment setup:
[image]
 
so i am looking to re-use saved segment sizes as a template and control each one as 11ch dmx fixture. :D but your code i think is not far away from my needs :D

mxklb, 2022-11-18:
@dzelionis the code shall not change any segment count nor segment size!

Configured segment size is not taken into account - this is the same as for existing [DMX_MODE_EFFECT](https://kno.wled.ge/interfaces/e1.31-dmx/). Every segment uses 13 DMX channels to be controlled, starting @ DMX start address as configured in E1.31 settings. So first segment is @ start address, next segment starts at start address + 13 .. `Segment DMX address = Start address + i*(13+x)`, where `i` is segment index and `x = dmxSegmentSpacing` as configured in UI. The tricky part is, I don't know how WLED segments are indexed during creation & deletion. I simple loop `first .. last` segments and apply each 13 channel values as received by E1.31 DMX data. You should try to setup your segments again, one after another with consecutive start..stop addresses. Or just find out once your 9 segment addresses by sending DMX. Who knows how segments get indexed ..

> @deggle Having said that, is there any harm in including it? 

Depends. Skipping disabled segments is like no action @ these segments 13 DMX channels and leaving a gap of `13 + dmxSegmentSpacing` channels. It is okay but may confuse or complicate usage. It may also be one more error users could make. I think it doesn't have much value for a DMX remote control - but it's a great way to handle segments within UI.
A benefit including it, if you have > 39 segments but just want to control some < 40 specific segments via DMX. 
I would consider to therefore implement multiple universe support to also get a more general up scaling.

Note: As DMX_MODE_EFFECT too, this only implements 1 universe with 512 DMX channels = 39 Segments :warning:

mxklb, 2022-11-18:
@dzelionis Normally DMX gear ist able to use any amount of DMX channels for any fixtures. So you may think about using proper E1.31 software to handle WLED DMX controls correctly. Sticking with 11 channels is not supposed to, just changing `dmxSegmentChannels=11` will lead to errors! This is a development issue, so I think you should be able read and understand my code changes. You should clearly understand whats implemented when reading the code! Sorry but sending segment index, segment size and 11 channels is not what is implemented here, please [RTM (Effect Mode)](https://kno.wled.ge/interfaces/e1.31-dmx/). Please try it with proper DMX soft-/hardware again with 13 channels and come back later if you think you found a bug. Sorry I'm not a support guy ..

deggle, 2022-11-18:
@max9901 @mxklb 

I'm trying to gauge the differences to see how we move forward. I suspect there is merit in merging the work into one, but we need to get some consensus on features vs. code changes and avoid things getting messy.

I correct that the only overlap here is in how the effect mode is implemented? I'm getting confused because there's mention of 11 vs. 13 - is the 13 just the whites or have you added additional channels into your version @mxklb?

mxklb, 2022-11-18:
As you see in code, what I implemented sticks to 13 channels. Last 2 are additional whites (as in DMX_MODE_EFFECT) but required. Actually this code part is copy & paste from DMX_MODE_EFFECT. In general you should be able to setup DMX fixtures with 13 channels, if whites are not supported by the stripe these channels simple do nothing.

I'll will merge @max9901 changes into mine or make a separate PR when I find time to ..

mxklb, 2022-11-18:
P.S with a little hack, `uint16_t` -> `int16_t` for `x=-2` and possible range [-2..150] one should be able to avoid last 2 white channels explicitly, so only using 11 channels. But a change like this is a hack and needs further testing, especially when reaching the end of the universe. Can someone explain to me how `availDMXLen` gets > 11?

deggle, 2022-11-18:
> As you see in code...

:)

I did check the code and it looked the usual channel layout, but the talk above of additional channels (both your comment of "So the only difference is ... that there are 13 channels per segment..." and the bit about sending segment indexes, which I think was just a misunderstanding) threw me a bit.

Anyway - yes, let's see if we can get these features pulled together and into PR.

@max9901 hopefully this works for you? If we can get this in, I have not forgotten that I did offer a £100/€100 bounty on this feature, which has your name on it for implementing my original (D)RGB(W) request.

mxklb, 2022-11-18:
I checked again. The `applyPreset(e131_data[dataOffset+0], CALL_MODE_NOTIFICATION)` part of `DMX_MODE_EFFECT` looks undocumented  as well and therefore it's a hidden feature, which triggers up to 256 presets @ DMX start address, when it is > 501. This means when setting start address accordingly `DMX_MODE_EFFECT` and `DMX_MODE_EFFECT_SEGMENTS` becomes a one DMX channel preset switcher. I've not tested it but I think this should be documented as a special behavior of these DMX modes - IMO more clean would be to therefore provide a new one channel DMX mode `DMX_MODE_PRESET` instead.. TBD, I think we shall do that refactoring too, what do you think?

dzelionis, 2022-11-18:
[> @dzelionis Normally DMX gear ist able to use any amount of DMX channels for any fixtures. So you may think about using proper E1.31 software to handle WLED DMX controls correctly. Sticking with 11 channels is not supposed to, just changing `dmxSegmentChannels=11` will lead to errors! This is a development issue, so I think you should be able read and understand my code changes. You should clearly understand whats implemented when reading the code! Sorry but sending segment index, segment size and 11 channels is not what is implemented here, please [RTM (Effect Mode)](https://kno.wled.ge/interfaces/e1.31-dmx/). Please try it with proper DMX soft-/hardware again with 13 channels and come back later if you think you found a bug. Sorry I'm not a support guy nor I have anything to do with soundswitch ..

@mxklb, sorry i know you not soundswitch support guy :D. Also,i was thinking that wled RTM [Effect Mode] in the documentation has  11 channels, far as i remember, last time i was checking it, it had 11...it may be that it was updated, so thanks for pointing that out.  My C language skills is not great (i use python for my needs),so i kinda get your code, but there are some gaps for sure, such syntax like this one:
`const uint16_t dmxLenOffset = (DMXAddress == 0) ? 0 : 1;` 
 i see the first time and i am not sure what value dmxLenOffset will get assigned. :D

Aircoookie, 2022-11-19:
@mxklb Thank you for your first implementation on this!
I think all three proposed DMX modes (Segment RGB, Segment DRGB, Segment effects) would be an useful addition.

mxklb, 2022-11-19:
All right, in actual state the `Effect` modes are divided into 11 & 13 channels with and without segment control ability, everything is implemented as a separate DMX mode. In addition `Presets` are separated as stand alone mode.

@Aircoookie I could not test presets because I don't have any presets and was not able to save some :disappointed: maybe there is a bug introduced within some of the latest changes from upstream, could please someone check it? I was able to save presets in past, maybe it's an issue of myself, I don't know ..

Expect `Presets` mode everything for `Effect` modes is finished and ready to be tested. Please give feedback.

To separate concerns of different features I already [pull request](https://github.com/Aircoookie/WLED/pull/2891) this.

dzelionis, 2022-11-21:
@mxklb , same issue here, can not save presets (still running your code). Onclick getting error:
 
[image]

i will try to check if the issue exists on the latest official wled version.

dzelionis, 2022-11-22:
@Aircoookie, fix for issue with Preset save:
need to modify index.js at line 2232  from:
```if ( gId(`p${i}lmp`).value!=="") obj.ledmap = parseInt(gId(`p${i}lmp`).value);```
to
```
                        if ( gId(`p${i}lmp`) !== null ) {
                          if ( gId(`p${i}lmp`).value!=="") obj.ledmap = parseInt(gId(`p${i}lmp`).value);
                        }
```
rebuild npm and save is working again...

max9901, 2022-11-22:
@mxklb  sorry I was a bit stale, but if u like I can also take the time to merge my changes into your PR shall I do that? and then open a new PR request that has both our functionality in there.

dzelionis, 2022-11-22:
> @Aircoookie, fix for issue with Preset save: need to modify index.js at line 2232 from: `` if ( gId(`p${i}lmp`).value!=="") obj.ledmap = parseInt(gId(`p${i}lmp`).value); `` to [...]
@blazoncek, maybe you have a quick check on this bug with presets, as i have compiled yesterday's latest version of wled and the issue was there, so i have found a solution for it too..you just need to apply that fix.....

blazoncek, 2022-11-22:
Fixed. Thanks for pointing it out @dzelionis .

dzelionis, 2022-11-22:
> Fixed. Thanks for pointing it out @dzelionis . [7135095](https://github.com/Aircoookie/WLED/commit/713509527a2591ae93ad22fce28c251fa6365199)

That no problem at all....

Thank you and @mxklb  did a very nice job, Thank you Mxklnb.

mxklb, 2022-11-22:
@max9901 I already provided a [PR](https://github.com/Aircoookie/WLED/pull/2891) for the `Effects` mode changes. I'll like to wait for feedback on this before merging our branches together. Nevertheless I already started reviewing some of the other DMX modes as existing in upstream. As far as I could read the code correctly, at least the modes  `DMX_MODE_SINGLE_RGB` and `DMX_MODE_SINGLE_DRGB` are likely to use one channel more than suggested/documented.

    wChannel = (availDMXLen > 3|4) ? e131_data[dataOffset+3|4] : 0; 

This line looks like it's using 4|5 channels instead of 3|4. So maybe this is an existing "bug" in implementation. TBD

    availDMXLen == Channels left in universe since start address != Number of DMX channels per Mode

I already cleared this out in [drgb branch](https://github.com/mxklb/WLED/tree/drgb), so maybe u may take that into account for the other DMX changes too ..

davidd1205, 2022-11-29:
hello
thank you for these improvements
I will also compile to test these new Dmx mode a wled to see how soundswitch reacts
@dzelionis but I think you have to ask Soundswtich to create another fixture for these new modes have you already asked? because with wled I get 2 fixtures

davidd1205, 2022-11-29:
I saw that the changes for SS were at the level of the effectoption and the R3, G3 B3
[bri,effectCurrent,effectSpeed,effectIntensity,effectPalette,effectOption,R,G,B,R2,G2,B2,R3,G3,B3]

on the other hand there is also a preset mode which seems interesting but not as such in a show but just for static it would be great to be able to call a preset with the effect mode is it possible to implement this? or maybe a realtime problem with wled

thank you in any case for this advance, for the moment I use the main segment with the group function to have a kind of multicell but without monopolizing the Dmx universe with few leds but the SS chases are not great for these strip pixel, I told them but he doesn't seem too concerned as we are a minority according to them to use this but I think this minority will grow

mxklb, 2022-11-29:
I think calling presets in DMX `Effect` modes is only needed rarely. Normally one would define different DMX states (called scenes) and trigger them instead. So you are actually already able to fire presets without even defining them in WLED :-)

As mentioned above `Presets` mode was already implemented some kind of .. If you try existing release in DMX `Effect` mode and choose DMX start address > 501, WLED presets shall switch (i did not test it but that's what was/is in code). Because this is uncommon practice for DMX addressing, it is moved into its own DMX mode in my implementation.

mxklb, 2022-11-29:
> @dzelionis but I think you have to ask Soundswtich to create another fixture for these new modes have you already asked?

That's correct! In most DMX software I know you're able to add user fixtures, with custom defined DMX channel layouts. Maybe better ask them to implement this instead. With the correct fixture mapping, any DMX software that sends E1.31 or Artnet is supposed to control WLED via DMX as described.

max9901, 2022-11-29:
> I think calling presets in DMX Effect modes is only needed rarely. 

@mxklb  this would however be a very nice way to control your segment settings. (just thinking out loud here)

blazoncek, 2022-11-29:
Please use WLED [forum](https://wled.discourse.group) or Discord for discussion.

mxklb, 2022-11-29:
https://wled.discourse.group/t/dmx-effect-mode-refactoring-beta-testing/6498

[44 more replies]