wled/WLED #2327: Can I use FastLED code in WLED?

matterdark11, 2021-11-09:
Hi,

I have another noob question: is there a way to use amination written for FastLED with WLED? 

Thanks a million for any help.

blazoncek, 2021-11-16:
You can. How? Not easy. 🤷‍♂️ 
WLED is based on NeoPixelBus which has a different approach than FastLED.

matterdark11, 2021-11-17:
thank you blazoncek. Basically to me it means it is not possible :)

closed 2021-11-17 by matterdark11

blazoncek, 2021-11-17:
I think there was a work from @embedded-creations or someone else that made porting FastLED code easier.

embedded-creations, 2021-11-17:
I have this on my out of date WLED branch:

https://github.com/embedded-creations/WLED/tree/FastLED_2D/usermods/FastLED_Compatibility

I need to merge the latest changes to WLED back into my branch, and will hopefully do that sometime after 13.0 is released