wled/WLED #2329: Poweroff delay

mase76, 2021-11-10:
Hi!
How can I add a generic poweroff delay? I want to add a delay of 10s when receiving a poweroff over sync.

Astaroth-OZ, 2021-11-20:
I am just a regular user of the application but with the transition settings you can achieve something like this, but keep in mind that the way I could do it, you will have the disadvantage that also when you start, you will have a delay of 10 seconds.

Config / LED Preferences / Transitions / Transition Time: 700ms

You just change the 700ms to "10000ms" = 10s.

I hope it will be of help.

mase76, 2021-11-20:
That is not the behaviour I want to have. With transitions I slso have a long fade in/out. I think, I have to modify the code for my usecase.