wled/WLED #2337: Color palette not working v0.13.0-b4

wgoldfish78, 2021-11-12:
### What happened?

When I select a palette nothing's happening.  Like for example f I select C9 it just stays the same color nothing happening. 

### To Reproduce Bug

select the palette c9 effect solid 

### Expected Behavior

from what I remember I would make a pixel of each color

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_v.0.13.0-b4 ''Toki''

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-11-12:
Not all effects use palettes.
Select an effect that uses palette.

Instead of Solid select Palette effect to display palette.

closed 2021-11-12 by blazoncek

wgoldfish78, 2021-11-15:
> Not all effects use palettes. Select an effect that uses palette.
> 
> Instead of Solid select Palette effect to display palette.

Thanks I guess that's how I did it last year