wled/WLED #2340: Add fading effect option for use as light alarm clock via Time-controlled presets

nikvostro, 2021-11-13:
Hi team!

I would like to use WLED as alarm clock by smooth fading in (of brightness) of selected preset for a certain schedule. Like it is already implemented for special Sunrise schedule. But I don't wan't to wake up with the Sunrises :) (ref. to Sunrise schedule in Time-controlled presets)

I suggest adding additional checkbox "fade in" for each schedule record. For fading effect (in term of coding) try to utilize effect for Sunrise with all it's seetings BUT with the preset specified in particular schedule.

The only alternative I see now is to setup several sequental schedules with the same preset and stepped increase of the brigthness.

Nikolay

blazoncek, 2021-11-14:
You can create a playlist which can do what you want.

nikvostro, 2021-11-17:
> You can create a playlist which can do what you want.

@blazoncek 
Thank you for advice! I've tried to implement it even 2 ways:
1. Transition between effects in the playlist -> maximum value is limited to 60 sec
2. Trasition for a single effect -> value is limited to 34.4 sec

But I want really slow sunrise for at least 5 or even 10 mins of fade out...
Hence these solutions doesn't solve my need...

blazoncek, 2021-11-17:
If you increment brightness by 1 every 60s this will give you rise time of 255 minutes.
Even with increment of 8 you are still in 32 minute rise.

Please look into APIs how to achieve this kind of playlist.

pyr0ball, 2021-11-24:
While this can be implemented using existing features, I think it's worth calling out that there are many people who use nothing but the out-of-box features with this application, and this feature is one that's highly desirable for a lot of people. It's been a feature on Philips Hue since like 2015.

Not saying that it's a necessity, but if some development time could be put toward implementing this as a GUI accessible feature, it would be a great boost to this project's usability

nikvostro, 2021-11-24:
> If you increment brightness by 1 every 60s this will give you rise time of 255 minutes.
> Even with increment of 8 you are still in 32 minute rise.
> 
> Please look into APIs how to achieve this kind of playlist.

I did achieve  my goal by creating 2 playlists + 2 presets:
1. First initial playlist to start the sequence from some effect. In my case I used just simple "Solid" with initial brightness 0. Then ending effect in the playlist will switch to the next playlist -> p.2
2. Playlist which is launching preset "increase brightness by 1"every 60 seconds.

But it would be much more comfortable if this scripting will be implemented as just one simple checkbox "fade in" near each row in scheduler :)

blazoncek, 2021-11-24:
Agreed on usability OOB. But someone has to code that (and test and document) while playlists offer you the flexibility to use any kind of combination.

blazoncek, 2024-01-02:
Effect transitions are implemented in 0.14 and above.
Combined with brightness transitions and playlists they allow for smooth fade-in/out.
Closing as resolved.
If you thimk the solution needs to be different please open a new issue with description that takes into account 0.14 behaviour.

closed 2024-01-02 by blazoncek