wled/WLED #2341: Controlling UserMod Relays through E.131

ideen1, 2021-11-14:
I have a custom-compiled version of WLED running on an ESP32 with 4 usermod relays enabled. I can set the pins for these relays and control them through the info screen on the WLED website with no problems. However, given that these relays are controlling traditional AC lights, I would like to be able to control them from XLights using E.131. Is it possible to do this?  Any help is appreciated.

blazoncek, 2021-11-14:
No. It is not.
Currently the only options to control relays are: MQTT, HTTP API, JSON API or hardware buttons.

ideen1, 2021-11-15:
Is this something that could easily be implemented? I think it would add a great deal of functionality, especially for lighting displays that also use some more traditional lighting

blazoncek, 2021-11-15:
Point me to specs and I'll see what I can do.

joerg74, 2021-11-15:
Hi, I`d also the idea to control a Relais by DMX, but due to my lack of knowledge of c ++ I was not able to  implement this.
I`m interestet in your current solution. To control your relays it should be possible to insert this in the class e131.cpp.
either each relay separate or the block of relays shoult have a configuration od the dmx channel. To power on the relais  any value more then 0 should work und 0  powers down.

ideen1, 2021-11-15:
How involved is the process to include a few hooks for relays so that they can be controlled by E131?

blazoncek, 2021-11-15:
Approval from _übermesiter_ himself is needed. 😀 
And since I do not know anything about E131 and have no specs on the protocol (I am lazy to google) I would need someone to add hooks into E131 so that either:
- E131 code would call usermod on a certain event (you may have different events each requiring separate hook) or
- usermod would be able to interrupt E131 processing (and decipher relay control) and decide what to do with the rest
The first option is preferred but requires more forward thinking.

ByronAP, 2021-11-17:
I used this code to control via dmx from xlights, pretty simple and easy to expand upon or change controllers. Down side is it requires using a separate controller for the relay(s), which for me actually worked out fine. I setup xlights to use white only on channel 0 universe 1;

#include <ESP8266WiFi.h>
#include <E131.h>
#include <ESP8266mDNS.h>

const char* ssid = "hgfdhfgh"; // your network AP SSID
const char* password = "gfhgfhgfh"; // your network AP password
const char* hostname = "strobe-relay";

//Set the universe and channel. Universe is E1.31 (DMX) universe.
//Channel will generally be 0, which is actually 1 (Red) in DMX. Use 1 for Green, and 2 for Blue.
const int universe = 1; // universe to listen to
const int channel = 0 ; // channel to use - these are 1 lower than normal (0 = 1)

int channel_val; // the current value of the channel we are using
long int num_ch_in_pkt; // number of channels in the recieved packet
int output_1 = 15; // names the ESP8266 GPIO output pin we are using (5 is D1 !!!)
E131 e131;

void setup() {
  Serial.begin(115200);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.setHostname(hostname);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    //Loop (forever) until WiFi is connected.
    delay(500);
    Serial.print(".");
  }

  if (!MDNS.begin(hostname)) { // Start the mDNS responder for esp8266.local
    Serial.println("Error setting up MDNS responder!");
  }
  Serial.println("mDNS responder started");
  
  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());
  
  //Set output pin and set it high (off).
  pinMode(output_1, OUTPUT); // set pin as an output
  digitalWrite(output_1, HIGH); // set pin to high
  
  e131.begin(E131_UNICAST, universe);   //Unicast (TCP)
//e131.begin(E131_MULTICAST, universe); //Multicast (UDP)
}

//Any DMX value under 127 is "Off", above 127 is "On".
void loop() {
  num_ch_in_pkt = e131.parsePacket(); // parse packet
  if (num_ch_in_pkt) {
    channel_val = (e131.data[channel]);
    Serial.println(channel_val);
    if (channel_val > 127) {
      digitalWrite(output_1, HIGH); // SSR on
    }
    else {
      digitalWrite(output_1, LOW); // SSR off
    }
  }
}

ideen1, 2021-11-18:
Thanks @ByronAP, I will give this a try on a separate controller. As for implementing this on the same controller, it seems to me that the E131 interface works by simply looping and waiting for packets from the DMX controller; Is it possible to insert this code somewhere into the WLED e131.cpp file to pickup on relay messages alongside the default led messages?

ideen1, 2021-11-18:
Does anyone have a good tutorial for compiling the @ByronAP code for an ESP32? I'm fairly new to the ESP/Arduino environment and seem to be having lots of difficulty finding the correct header files for it to compile successfully.

ByronAP, 2021-11-19:
@ideen1 here ya go.. ESP32

```c++
#include <WiFi.h>
#include <ESPAsyncE131.h>

const char* ssid = "Penner-2.4G"; // your network AP SSID
const char* password = "llaannee#1979"; // your network AP password

//Set the universe and channel. Universe is E1.31 (DMX) universe.
//Channel will generally be 0, which is actually 1 (Red) in DMX. Use 1 for Green, and 2 for Blue.
const int universe = 1; // universe to listen to
const int universe_count = 2; // Total number of Universes to listen for, starting at universe
const int channel = 0 ; // channel to use - these are 1 lower than normal (0 = 1)

int output_1 = 15; // names the ESP32 GPIO output pin we are using for our relay (SSR)

ESPAsyncE131 e131(universe_count);

void setup() {
  Serial.begin(115200);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    //Loop (forever) until WiFi is connected.
    delay(500);
    Serial.print(".");
  }

  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());

  //Set output pin and set it high (off).
  pinMode(output_1, OUTPUT); // set pin as an output
  digitalWrite(output_1, HIGH); // set pin to high

  e131.begin(E131_UNICAST, universe, universe_count);   //Unicast (TCP)
  //e131.begin(E131_MULTICAST, universe, universe_count); //Multicast (UDP)
}

//Any DMX value under 127 is "Off", above 127 is "On".
void loop() {
  if (!e131.isEmpty()) {
    e131_packet_t packet;
    e131.pull(&packet);     // Pull packet from ring buffer

    uint packetUniverse = htons(packet.universe);

    if (packetUniverse == universe) {
      uint packetValue = packet.property_values[channel + 1];
      if (packetValue > 127) {
        digitalWrite(output_1, HIGH); // SSR on
        Serial.println("SSR ON");
      }
      else {
        digitalWrite(output_1, LOW); // SSR off
        Serial.println("SSR OFF");
      }
    }
  }
}

ByronAP, 2021-11-19:
@ideen1 here ya go.. ESP32

```c++
#include <WiFi.h>
#include <ESPAsyncE131.h>

const char* ssid = "hkhkhkkhjkh"; // your network AP SSID
const char* password = "jhjkh"; // your network AP password

//Set the universe and channel. Universe is E1.31 (DMX) universe.
//Channel will generally be 0, which is actually 1 (Red) in DMX. Use 1 for Green, and 2 for Blue.
const int universe = 1; // universe to listen to
const int universe_count = 2; // Total number of Universes to listen for, starting at universe
const int channel = 0 ; // channel to use - these are 1 lower than normal (0 = 1)

int output_1 = 15; // names the ESP32 GPIO output pin we are using for our relay (SSR)

ESPAsyncE131 e131(universe_count);

void setup() {
  Serial.begin(115200);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    //Loop (forever) until WiFi is connected.
    delay(500);
    Serial.print(".");
  }

  Serial.print("IP address:\t");
  Serial.println(WiFi.localIP());

  //Set output pin and set it high (off).
  pinMode(output_1, OUTPUT); // set pin as an output
  digitalWrite(output_1, HIGH); // set pin to high

  e131.begin(E131_UNICAST, universe, universe_count);   //Unicast (TCP)
  //e131.begin(E131_MULTICAST, universe, universe_count); //Multicast (UDP)
}

//Any DMX value under 127 is "Off", above 127 is "On".
void loop() {
  if (!e131.isEmpty()) {
    e131_packet_t packet;
    e131.pull(&packet);     // Pull packet from ring buffer

    uint packetUniverse = htons(packet.universe);

    if (packetUniverse == universe) {
      uint packetValue = packet.property_values[channel + 1];
      if (packetValue > 127) {
        digitalWrite(output_1, HIGH); // SSR on
        Serial.println("SSR ON");
      }
      else {
        digitalWrite(output_1, LOW); // SSR off
        Serial.println("SSR OFF");
      }
    }
  }
}

joerg74, 2021-11-23:
wouldn`t it a good idea to call a Method in the usermod implementing class? something like dmxCall (universe, channel, value)

cmgreenman, 2021-11-29:
Following as I am trying to drive a Sainsmart relay board over e1.31 as well.   Everything works in the app but when playing an e1.31 stream all 7 channels play on the first 3 strings.

broncosis, 2022-02-25:
@blazoncek we chatted on discord a while back about the same thing 
I think count me in for any beta testing I have several D1 minis and relay boards 
willing to make the ultimate sacrifice for this.

ZixZeven, 2022-06-21:
HI, I am using the above ESP32 code to control 1 relay on xLights with success.  My questions is: it possible to control multiple relays with this sketch?  What do i need to change?

blazoncek, 2022-08-01:
https://wled.discourse.group/t/xlights-controlled-relay/5139

closed 2022-08-01 by blazoncek