wled/WLED #2342: Support Chip LED: SK6812 IC

pepsonEL, 2021-11-17:
Hi. I buy this christmas LED Strip and one question...WLED support it ?
Chip LED: SK6812 IC

https://pl.aliexpress.com/item/1005002247063442.html?src=google&src=google&albch=shopping&acnt=494-037-6276&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=6459787116&albag=77066918093&trgt=1463104581777&crea=pl1005002247063442&netw=u&device=c&albpg=1463104581777&albpd=pl1005002247063442&gclid=CjwKCAiA7dKMBhBCEiwAO_crFAa3YLmzL2g6eI6mdow4u-Wrhanzj45W3jXqYLCzHRCMO3B4tFETlRoCJdAQAvD_BwE&gclsrc=aw.ds&aff_fcid=11914cc1f3c746d48787984786f024fb-1637146104660-02635-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=11914cc1f3c746d48787984786f024fb-1637146104660-02635-UneMJZVf&terminal_id=4561427e85ef4cab8613564132edae93

Aircoookie, 2021-11-17:
Yes.

closed 2021-11-17 by Aircoookie

pepsonEL, 2021-11-17:
thanks. I start testing :)

LocalNic, 2021-11-30:
I have exactly these Strips in a Project, but I still have some difficulties with those.
I use these sk6812 on a Wemos Mini d1 ( pin D2 ) and for verification on a  SP110E (which  comes with the strip)
The SPE110 works fine and shows all colors including the white channel.
For me it seems that the Register is shifted, when I turn on a solid red, then the strip is showing 1st = Red/  2nd Green / 3rd Blue / 4th = off 
When I additionally turn white on the 4th LED just turn Red in the appropriate Level of the White channel. 
The effects are completely strange and look random obviously. 
It also seems that I need to configure a higher LED number to address the whole strip. 
First I consider a voltage Problem or a defect LED but the effects were too consistent and the verification with the SP110E afterward shows a perfect working LED string. 

Hope someone can help

blazoncek, 2021-11-30:
Did you set SK6812 as the LED type in settings?

LocalNic, 2021-11-30:
yes I have and also change the RGB or GRB settings but lead to the same result.

blazoncek, 2021-11-30:
Please post pictures of your LED settings and actual LEDs (closeup).

LocalNic, 2021-12-02:
Hi, 
now I managed to take some pictures and also verified if the string works and has no dead pixels. 
With an SP110E everything works fine in the Software  I configure an SK6812 String with 40 LEDS ( even if the String only have 20 Pixels) 

So I did each color by itself with an WEMOS D1 mini with the actual WLED firmware. connected to GPIO 4 which is named D2 on the device.
Power comes from the 5V VCC of the WEMOS smoothed with a small capacitor. DataLine has a 68Ohm Rasistor between the string and the Wemos. 

There is one config strange. The String have 20 LEDs but I need to configure a LED range between 20-40 to light up the whole string. 
from 0-20 no LED lit up 
everything above add 1 led on the string but shows the color shift. 

only white channel looks like:
[image]

only red looks like :
[image]

only green looks like : 
[image]

only blue channel looks like : 
[image]

WLED config are screenshot from mobile, hope I capture all necessary 
[image]

Hope this helps. a real close up of a LED you can see in the aliexpress link above 
I hope as well that I have explained it most accurate ;)

blazoncek, 2021-12-02:
This doesn't look like SK6812 to me.
Use WS281x type.

LocalNic, 2021-12-02:
OK, did what you suggest ! 
And it works .... well partly .... I need to set it to WS2812 ( for some reason it is sold as SK6812) but I need to configure the segment pretty unusual. So I need to configure 40LEDs while having a 20 LED String and set the starting LED to 20 . 
Then the Effects of WLED are working pretty fine . 
Thanks for the quick and awesome help !

blazoncek, 2021-12-02:
It may be due to LEDs having fixed addresses.

darkmattercoder, 2022-02-21:
#1640 #1312 seem to be related

darkmattercoder, 2022-02-21:
I am not sure. My one strip currently under test seem to have baked in addresses

60 to 77 ( I cut two off at the end)

However, the settings for start, length and segment setting seems to be weird. I do not yet fully understand whats going on. 

But here are a few examples:

## Situation
One strip connected on pin 2. Physically there are 18 LEDs in it. I did cut 2 leds away in the end no matter why. From my findings, the baked addresses of those leds are 60 to 77. Here is why I got those numbers. All analysis has been made in wled 0.13.0-b6_ESP32. Led type set to WS281x, RGB.

1. start `0`, length `60` -> no leds are lit
2. start `0`, length `61` -> first led is lit
3. start `0`, length `77` -> all leds except the last one are lit
4. start `0`, length `78` -> all leds are lit

Well, so as we now know th led addresses, we can configure the strip correctly, right? Therefore we check "custom bus start indices".

1. start `60`, length `18`: 
  * No leds are lit at all
  * Segment 0 automatically created is Start `60` stop `78`
  * Reset segment leads to segment 0 being Start `0` stop `78` 
  * Segment settings have no effect, leds stay off
  * LED Preferences top area reads `Total LEDs: 78 (18 physical)`

  This all looks a bit weird to me. Total LEDs 78 is a conclusion which cannot be made when only one output with 18 leds has been defined, regardless of the start index. The control package should contain data for 78 leds, however, the description is somehow unspecific on that. Would better read `Total led datagram count: 78 (18 physical leds)` ore something. Segment settings seem to be not respecting the sequential led settings when a start index different from zero is used. I have no good solution here beside completely decoupling the `physical adresses` of the leds from the `order of the leds` and introducing an additional abstraction layer for configuration. The whole thing about *strips containing hardcoded addresses* is a nightmare from an UX. I think  it would be a really tough task to address (pun intended) both setups from the same (WS281X) protocol without proper decoupling. There should be an entirely new protocol (WS281X with fixed addresses) to make them really distinguishable.

2. Start `60` length `78` -> just being curious:
  * All leds are lit
  * Segment 0 automatically created is Start `60` stop `138`
  * Segment settings to make first led lit up -> start `120` stop `121`
  *  Segment settings to make first last lit up -> start `137` stop `138`
  * LED Preferences top area reads `Total LEDs: 138 (78 physical)`

There seems to be some weird behaviour here. Overall. With the settings above and the segment being set to start `120` stop `138`, I can work with it.  Alternatively, I could achieve the same by using start `0` and length `78` and using the segment from `60` to `78`.

I cannot say more until I get a scope hooked onto it. But I think that all the above clearly shows at least some edge case bug in wled when working with fixed-address led strips.

darkmattercoder, 2022-02-22:
Without having really any time to dive into the code, from my first ever glance into the wled code, the following line *might* prevent the *start `60` length `18`* situation to work:

https://github.com/Aircoookie/WLED/blob/565d8d8f04f1d321dd3c92914400929ecdef1348/wled00/bus_manager.h#L513

Here the setColor method immediately returns when the requested pixel is higher than `_len`. I still might be wrong, when somewhere else the value of `pix` is preprocessed based on how the bus start index is set. 

I would love to really hook my debugger into this to hunt that issue down, however before the end of march I cannot put any time in it. Maybe @Aircoookie is able to do a quick fix, knowing the code the best.

Aircoookie, 2022-02-22:
@darkmattercoder Bus start indices don't work this way. They allow you to set physical LED 1 as if it was WLED LED 60, not set physical LED 60 as WLED LED 1.
There is no native support for skipping the first x LEDs of a bus (apart from "Skip first LED").
The easiest way to address your LEDs correctly is to set the bus as start `0`, length `78` and then go into the main UI and set Segment 0 to reach from `60` start to `78` stop. In order to keep that setting after a reboot, make a preset and save it as Boot preset in LED settings :)

darkmattercoder, 2022-02-22:
Thankls for the reply,

yes that is exactly what my finding is. But are you willing to accept a general pull request which addresses this? e.G. Are those fixed address strips something you want to have as valid strips in Wled with a more user friendly abstraction in the interface? This however would most likely require a lot of additional work. If you want to make this abstracion solid and nice, I think that basic approaches have to be changed. Do you feel more like those things are out of scope for what Wled should handle and what not?

Aircoookie, 2022-02-22:
On a technical level, just replacing the "Skip first LED" checkbox with a number input to skip `N` LEDs instead would likely be enough to make this more user friendly. The logic for skipping >1 LEDs is otherwise already present in `bus_manager.cpp`.
Therefore, I don't think it requires a lot of extra work or would be outside the scope of WLED :)

BugsBunny1403, 2022-03-15:
> On a technical level, just replacing the "Skip first LED" checkbox with a number input to skip `N` LEDs instead would likely be enough to make this more user friendly. The logic for skipping >1 LEDs is otherwise already present in `bus_manager.h`. Therefore, I don't think it requires a lot of extra work or would be outside the scope of WLED :)

It's an good news for support this strip. It is possible to create an other option like WS281x or SK6812 to authorize this logic to run with this strip ?

Aircoookie, 2022-03-15:
This capability would be available to all digital busses :)

BugsBunny1403, 2022-03-15:
Nice, you have an idea of deploying or testing because i'm work on project with this strip. So i'm very interested

darkmattercoder, 2022-03-15:
I described that above and what in my opinion was required to entirely change it. But from the answers I consider that this is an undesired fix because to truly get this properly abstracted the current bus has to get refactored to something else which it has not been designed to be. The current interface, segment config and skipping of n LEDs does not fit the different approach that in my opinion is needed to integrate this. It can be used as is to drive those kind of strips perfectly fine. But it won't feel good from an user experience point of view.

blazoncek, 2022-03-15:
You misunderstood.
All logic for skipping N pixels at the beginning of a strip is already built in. It is just limited to 1 sacrificial pixel.
What you need to do is:
- compile yourself by changing LED_SKIP_AMOUNT in FX.h (to number of unavailable pixels)
- or updating settings_leds.html and set.cpp to include numeric field that will allow you to enter appropriate number

It is just something nobody really needed until now.

Aircoookie, 2022-03-16:
Made checkbox a number input instead as I think its useful for these fixed address lights.
Doesn't add extra overhead so why not, can now skip up to 255 pixels.
(and as an added bonus, memory use for skipped pixels is now correcly calculated)

BugsBunny1403, 2022-03-16:
Can you release this modification ?

blazoncek, 2022-03-16:
You can use Discord bot to compile from latest source.

BugsBunny1403, 2022-03-16:
Okai, i've not know this functionnality .

darkmattercoder, 2022-03-16:
> You misunderstood. All logic for skipping N pixels at the beginning of a strip is already built in. It is just limited to 1 sacrificial pixel. What you need to do is:
>
> * compile yourself by changing [...]

But I do not see from the code, that this will affect the segment appearance as I described above in my longer answer, or does it and I did not see it?

darkmattercoder, 2022-03-16:
Ah plus: what if you want to skip leds in between? e.g. chaining led 2-45 and 77-10 to one single segment on the same bus?

blazoncek, 2022-03-16:
> Ah plus: what if you want to skip leds in between? e.g. chaining led 2-45 and 77-100 to one single segment on the same bus?

create a custom mapping. unfortunately you are pushing it a bit in this case.
I will lock this issue since it is slightly abused.
Use WLED [forum](https://wled.discourse.group) or [Discord](https://discord.gg/WUpzKMaUKX) for help and support questions.