wled/WLED #2345: Factory reset for inaccessible controller

prabbit237, 2021-11-18:
**Is your feature request related to a problem? Please describe.**
Sometimes the LED controller may be installed in such a way that it's not easily accessible (if at all) for reflashing. So if something really got messed up in the settings and you couldn't access it to reflash and if it's not connecting to wifi or had a password set, etc. you might not be able to fix it. I am making some gifts where I want to bury the esp8266 inside of a mood-lamp and have it only accessible/operational via wifi or IR remote. But if someone is playing in the settings and doesn't know what they're doing and locks themselves out, I want a way that they can easily reset the whole thing.

**Describe the solution you'd like**
Tasmota (and some other firmwares) can be factory-reset by toggling the power repeatedly and with short uptimes as explained in the following steps: 

1) Cut power from the device completely for 30 seconds.
2) Power the device on and off six times with intervals lower than 10 seconds and leave it on after seventh time.
3) Fast power cycle device recovery should activate and the device should be reset to firmware defaults.

Something like this would be very helpful for making it more fool-proof.

tarontop, 2021-11-19:
Athom’s equipment seems to be there
https://www.aliexpress.com/item/1005003512212258.html?spm=a2g0o.store_pc_groupList.8148356.3.3d6546420DNjpx

Aircoookie, 2021-11-20:
Hi, for now I would recommend incorporating a button into your lights, pressing that for 7 seconds would open WLED-AP with the default `wled1234` password.

The repeated power cycle method sounds quite good - of course it would require a write to the filesystem twice every boot to increase the "fast power cycle" counter (and reset it after 10 seconds), which might not be good for flash endurance in the long run, but I might consider adding it regardless. If 5-6 cycles are required, it is also unlikely for it to reset unintentionally, e.g. in case of an unreliable power supply.

blazoncek, 2021-11-20:
A similar approach (albeit a bit less reliable) would be to add another button feature if button is pressed for more than 10s (or appropriately more) and reset settings.

prabbit237, 2021-11-20:
For my use-case, a button wouldn't be feasible. I've found some vases at the $1 store and pillar candles that fit in the top nicely. They also sell foam beads that are like Styrofoam except they're not melded together. I drilled a couple of tiny holes in the bottom for power wires to come out to a jack but a column of LEDs and D1 Mini will be inside with the foam beads poured in. The beads diffuse the lights nicely. Then the top is sealed about 1" in so the candle can sit in the top. Will make some nice gifts for Christmas.

I found the code that does the factory reset so I think I can add this to the userloop() for myself without too much issue. As to the "wear out the flash" part, it could be made as an optional feature for compile-time.

And since I'm making these for folks who won't necessarily know how to re-configure them, thus needing them fairly foolproof, and I'm using some of those new, cheap LEDs as mentioned in [https://github.com/Aircoookie/WLED/issues/1312](https://github.com/Aircoookie/WLED/issues/1312), I also had to figure out how to hard-code an off-set into the system (Yes, it can be done by segmenting them but then the person needs to know *HOW** to do it as well as the offset for that specific lamp. But I setup a compile-time define with the offset needed for any given lamp so it's baked in and survives a factory reset.)

richard-scott, 2022-02-10:
Maybe you can get some ideas from how Tasmota do it?
https://tasmota.github.io/docs/Device-Recovery/

PatrickHuetter, 2022-09-02:
+1 for this feature. An implementation like tasmota did it would be very nice! https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery As far as i know the Shellys are also doing it this way.

blazoncek, 2022-09-02:
Similar thing has been implemented 6 months ago for devices that have a button.
https://github.com/Aircoookie/WLED/commit/1bc698ae78dfadeade6143983005815042e972dd

It is easy to incorporate a button on any WLED project (which may not be the case with Tasmota).

richard-scott, 2022-09-02:
With respect @blazoncek, most of my WLED installations are in hard-to-reach places, where having a button is impractical.

blazoncek, 2022-09-02:
**All** of my installations, including the ones in hard to reach places, have button.
**It is not impractical and does not add excessive cost or bulkiness to the set up.** And I do have 30 around my house, ranging from ESP01 to ESP32 in Dig-Quad.
You may not have noticed (since we try to keep flash writes to a minimum) but flash wear is a problem, especially on cheap ESP clones.

PatrickHuetter, 2022-09-02:
I can only agree to @richard-scott . It's not always possible to have a button. In my case i'm having a lot of controllers flush-mounted and can't add a cover that can be easily opened because it wouldn't look nice anymore. Since this feature could be easily optionally activated there shouldn't be any problem for anyone or am i missing something here? There could be a boolean flag in the settings that could be activated to have this feature enabled. Or a different solution would be to have recovery mode while boot option which checks wether a specific WLAN SSID exists and if yes, connects to that. So in case of emergency you could just create a new Wifi (or access point) with a specific SSID name and re-power the ESP. After that it will connect to the emergency wifi and it would be possible to re-configure the WLED Controller.

Another thing is: Most of the ready to use available controllers don't have the possibility to re-arange or disassemble the button to get it outside somewhere. So in my opinion this would be a very practical and nice feature which also can be enabled optionally to prevent too many flash writes or just use another procedure as i described in the mid part of this answer.

blazoncek, 2022-09-02:
Do not get me wrong, I agree with you on the usefulness.
Still, someone need an incentive to write something for OSS. It is of no use to me so it is on low priority list.
If you have C/C++ knowledge you can try to implement it yourself. It should go into wled.cpp (setup() function)

Aircoookie, 2022-09-02:
Would also make a good usermod.

blazoncek, 2022-09-02:
Agreed. An there is no better place to start as Auto save usermod.

richard-scott, 2022-09-04:
Do user mods survive OTA firmware updates?

blazoncek, 2022-09-05:
Yes, but only if binary you are uploading has been compiled with usermod enabled.

richard-scott, 2022-09-05:
What about via the Home Assistant initiated update?
I’m assuming that would come direct from the internet, so for most that makes user mods a bit useless?

Aircoookie, 2022-09-05:
Yes, Home Assistant would only flash the vanilla (no user mods) binary, to keep the user mods you need to do the update right from within WLED with the custom .bin that contains all usermods you want to have.

richard-scott, 2022-09-05:
Is there a way to either get HA to send a URL to flash? or pre-set the OTA URL on the device before HA asks for it to update?

szilardcsotya, 2023-01-12:
> **Is your feature request related to a problem? Please describe.** Sometimes the LED controller may be installed in such a way that it's not easily accessible (if at all) for reflashing. So if [...]
Bought an WLED-controller with IR remote from Aliexpress:

https://www.aliexpress.com/item/1005004929457839.html?spm=a2g0o.productlist.main.1.3c85jYOQjYOQUE&algo_pvid=de8ccee0-8596-49ae-bed3-b58c196a4c34&algo_exp_id=de8ccee0-8596-49ae-bed3-b58c196a4c34-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000031056110793%22%7D&pdp_npi=2%40dis%21HUF%2117474.8%2112231.22%21%21%21%21%21%402102160416735499986146937d06cd%2112000031056110793%21sea&curPageLogUid=b2TdD2rabEnV

After I set up the controller with the led strip everything was ok. After 5 mins the controller lost the signal of the IR remote, could not operate with remote anymore. There is a button at the top of the controller, pressing more than 7 sec is resetting the controller, I tought it would solve the problem. When the controller was working I set up the wifi AP on the controller with no PW. After rebooting I can see only the WLED-AP as a new wifi network not the WLED-SoundReactive as it was at the first time. Tried the resetting method again, tried the 6times power on-off, nothing solved the problem, the controller did not go back to factory settings. What am I doing wrong or what should I do? Thanks (Sorry for the huge link).

blazoncek, 2023-01-12:
@szilardcsotya Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

blazoncek, 2023-01-12:
@prabbit237 I have created a usermod with feature you requested.
Available in `power-ap`branch.

szilardcsotya, 2023-01-12:
> @szilardcsotya Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

Sorry for the wrong topic, thanks for the link. BB

plbeaudoin, 2023-03-25:
I'll share this information here if anyone ends up here like me because they're unable to reach their device that has no button (i.e. an Athom bulb).  WLED-AP will be accessible again if it can't connect to the configured AP.  I was able to access my bulb again by turning off my router (therefore making the configured SSID unavailable).

prabbit237, 2023-03-27:
> I'll share this information here if anyone ends up here like me because they're unable to reach their device that has no button (i.e. an Athom bulb). WLED-AP will be accessible again if it can't [...]

And that's great if the settings aren't screwed up by someone setting a password to lock the settings and then forgot the password or something such. Just remember; "if something is fool-proof and unbreakable, there's some fool that will break it."

blazoncek, 2023-03-28:
As noted above, several power cycles to reset device into AP mode has been implemented as a usermod in the `power-ap` branch.

It may get imported into main branch at some point. Until then use it by self compiling.

closed 2023-03-28 by blazoncek