wled/WLED #2348: IR Remote 24-key "ON" does not work

Janideluxe, 2021-11-20:
### What happened?

I connect a IR Diode and want to remote controll my stripe but the "ON" Button doesnt work. Everything else works as well (off brightness colors etc.). I receive the ON Code with an arduino and its exact the same 0xF7C03F as in the json

### To Reproduce Bug

if i push off i can't pull it on with IR remote. every time

### Expected Behavior

want to pull the stripe on with IR remote.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.13.0-b5_ESP8266.bin

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-11-20:
Post your LED settings page showing IR set-up.

Janideluxe, 2021-11-20:
[image]

Janideluxe, 2021-11-20:
ok i think i solved the problem by changing the GPIO Pin

closed 2021-11-20 by Janideluxe

blazoncek, 2021-11-20:
Best GPIOs for IR on ESP8266 are GPIO12, GPIO13 and GPIO14

Janideluxe, 2021-11-20:
thx for your help @blazoncek

Aircoookie, 2021-11-20:
Actually this problem was caused by GPIO 2 always being forced HIGH to turn off the onboard LED when the LED strip is off. It also did that when something else (like IR in this case) was configured to use GPIO2. This is now fixed in d31e4c78 so that GPIO2 is only set HIGH if there is a digital bus on the pin and no off refresh is required.