wled/WLED #2349: Trigger E1.31 "Live" from Home Assistant

ipodmusicman, 2021-11-21:
I want to create a party button that when pressed, will turn off all lights, start the music and get my LEDs to dance to LedFX, but in order to do this, I need to get Home Assistant to tell WLED to "turn on" E1.31 i.e. HA to do what I currently do when I "click on the star" so that LedFX can take over WLED.

I could not find a way to do this.  Any chance that this feature can be considered for the next version of WLED?

sansillusion, 2021-11-21:
In my configuration.yaml I have :

    switch:
        !include switch.yaml

To include switch.yaml and in switch.yaml I have :

    - platform: rest
    name: WLED Hyperion Toggle
    resource: http://192.168.1.162/json/state
    body_on: '{"lor": 0}'
    body_off: '{"lor": 1}'
    is_on_template: '{{value_json.lor == 0}}'
    headers:
      Content-Type: application/json

Hope it helps. Works great for me.

ipodmusicman, 2021-11-21:
Ah great! Thanks very much.  Worked for me too.

blazoncek, 2021-11-22:
Why not leave E1.31 turned on all of the time?
When WLED does not receive live data for more than 2.5s it will revert to whatever it was doing prior to receiving live data. So when you stop LedFx it will revert to playing preset/effect.