wled/WLED #2350: Playlist return to original feature

EnzoMaitrix, 2021-11-21:
Hi,

I have a installation with a infrared sensor as switch so I can start a playlist.

The playlist start and play correctly.

I will like to be able to return to the original preset at the end of the playlist (from where it begin before the sensor trigger)

Saving the preset as current preset or original preset and call him at the end of my playlist.

Also if it possible that the sensor/switch are visible in Home Assistant so I can have a log when it is trigger

Thank you for making the world brigther!

blazoncek, 2021-11-22:
All of this available in my fork for testing.
Use @srg74 repository for binaries. You will need bin with PIR usermod enabled.

blazoncek, 2021-12-26:
Closing since no feedback.

closed 2021-12-26 by blazoncek