wled/WLED #2351: Installation via EPSInstaller

RomeHein, 2021-11-22:
**Is your feature request related to a problem? Please describe.**
I had some trouble to install WLED even via ESPHome

**Describe the solution you'd like**
Some times ago I made [ESPInstaller](https://github.com/RomeHein/ESPInstaller). It's an Arduino firmware you can install directly via the Arduino IDE without any dependencies. It gives you that possibility to install via different ways:
- locally:  any .bin you may have
- remotely: with an URL to the .bin
- directly from the repo list if the repo has ESPInstaller config [files](https://github.com/RomeHein/ESPInstaller#how-can-i-add-my-project-to-the-repo-list) (basically a json file describing all versions URLs.

**Additional context**
I thought this could be a nice "installation way" in additions to those already available, as you don't have to play with command tools, but only Arduino IDE.

Let me know what you think about the idea ;)

Aircoookie, 2021-11-22:
Hi!
Thanks for reaching out to me and thank you for creating such an awesome project!
I doubt ESPInstaller will beat https://install.wled.me/v2.htm in ease of use for most, but I still believe that it is a great alternative to other USB flashers and seems better than all other OTA updaters I've come across!
I'd love to add WLED to the projects supported by ESPInstaller :)

Will test it and make a PR to your repo! One hurdle might be the partition sizes, we use a custom partition table for WLED now since 0.13.0-b5 (1.5MB app, 1MB filesystem). With the default (1.2MB app, 1.5MB filesystem) we are at 98% flash utilization, so future versions are likely to no longer fit 😅 On the other hand, the 190kB FS provided by the minimal spiffs scheme is a bit too small for a user who would like to make a lot of presets, so I need to decide.

RomeHein, 2021-11-24:
Oh I didn’t know about https://install.wled.me/v2.htm! That’s pretty cool! can’t beat it for sure 😅
For the partition size, it might not be clear in the project readme, but you can use what ever you want to install ESPInstaller. Actually, you need to set the partition size for the project you are aiming for. So in the case a WLED, users would have to choose 1.5MB app, 1MB file system prior installing ESPInstaller. That’s actually a feature I’d like to add in ESPInstaller, running a partition size check before running the OTA, and notify the user if it’s not right.
Let me know if you have any suggestions, anything I could/should improve. And thanks for your answer!

closed 2022-02-04 by blazoncek