wled/WLED #2352: WLED uses NeoPixelBus but does not mention it anywhere in the readme or official website.

Makuna, 2021-11-22:
Please be fair and mention ALL libraries you rely on.

Aircoookie, 2021-11-22:
I am very sorry about that.
Of course I credited NeoPixelBus in https://kno.wled.ge/about/contributors/#used-libraries-and-dependencies (in fact even the first entry after the arduino core), but it seems like while porting from Github Wiki to the MkDocs page a line break went missing and it was accidentally put in the line above.
Also it linked to the now long outdated fork by svenihoney, I've updated the link the link to your main repo.

Will also improve the link within the readme to point to above knowledge page rather than the outdated wiki.

Aircoookie, 2021-11-22:
Readme updated as well and outdated wiki credit removed.
Once again I owe you a sincere apology for not keeping the credits page updated sufficiently.

closed 2021-11-22 by Aircoookie