wled/WLED #2353: Can you upload firmware? I can't compile

Awe7, 2021-11-23:
I'm going to conclude this for the time being.

Support for the ESP32-SOLO-1 / WiZ Analog LED Strip is possible, but until this ESP32 variant is better supported in core, one must make some modifications to get it working.

[image]

[image]

In conclusion, the following steps are required to get WLED running on this hardware.

1.  Swap some libraries out: https://github.com/lbernstone/arduino-esp32-solo to support the single core ESP-SOLO-1
2. Add in a `delay(1)` at the end of the main loop: https://github.com/Aircoookie/WLED/blob/1d4487b6cd7c9a69bbce45c14ae1fb1c622e1d0e/wled00/wled.cpp#L121
Otherwise the ESP won't be able to multitask.
3. Attach to the ESP32-SOLO-1 RX/TX/GND pins for serial communication and Flashing.
4. Bridge IO0 to GND before booting to to being flashing process
5. Compile for ESP32 (with the aforementioned library swap)
6. Cross your fingers and Flash
7. Remove IO0 from Ground and power cycle
8. WLED should boot if successful.
9. Set LED output type to `PWM RGBWC` with pins 25, 26, 27, 4, 2
10. Set Button 0 pin to 16
11. Enjoy!

_Originally posted by @ChuckMash in https://github.com/Aircoookie/WLED/issues/2171#issuecomment-917262843_

Aircoookie, 2021-11-23:
I don' think you meant to make this a new issue :) Your comment in #2171 will be kept.

closed 2021-11-23 by Aircoookie

Awe7, 2021-11-23:
No, I can't