wled/WLED #2355: ruban LED WS2811 + WLED + SONOFF

arnaud-aub, 2021-11-23:
Bonjour, je reviens vers vous après un grand nombre de test et je ne comprends pas, WLED sur SONOFF fonctionne avec des pixels WS2811 mais pas avec un ruban led WS2811, sur le ruban, les led ne fonctionnent pas correctement pourquoi ? Merci

PS: j’ai déjà modifié tous les paramètres et rien ne change.

arnaud-aub, 2021-11-25:
Bonjour, j'ai essayé une énième fois mais rien ne change, quelqu'un peut m'aider ?

sansillusion, 2021-11-25:
Comment allimente-tu ton ruban. Il faut que le "ground" soit connecté entre les LED et le sonoff pas juste le signal. Le sonoff n'as pas assez de pouvoir pour allimentez le ruban, il faut donc un power supply.

arnaud-aub, 2021-11-25:
Oui je connecte bien tout ce qu'il faut, ainsi qu'une alimentation à part, tout fonctionne avec des ws2811 en pixel mais pas en ws2811 en ruban sans changement, je ne comprends donc pas d'où vient le problème.

arnaud-aub, 2021-11-29:
Bonjour, n'avez-vous pas d'aide à me donner ?

sansillusion, 2021-11-29:
Peut être que le ruban est défectueux ? Vous utilisez biens le data in (le bon bout du ruban) ?
Peut être que la première del est défectueuse, j'ai déjà endommagé une bande et après avoir retiré les deux premiers DEL le reste fonctionnais bien.

arnaud-aub, 2021-11-29:
Non, le problème ne vient pas du ruban avec un programme déjà tout fais sur Arduino tout fonctionne à la perfection, tout est correctement branché, je ne sais pas d'où vient le problème mais en tout cas ni du ruban ni de l'alimentation ni du branchement, mais merci quand même

jzeeman01, 2021-12-16:
Arnaud
Not sure this is the same...
If you have an external power supply - I connected the wire I soldered to the GRD pin to the earth on the power supply and also only got flashing lights....try connecting the wire to the V- on the power supply and not GRD - this worked for me

closed 2022-04-27 by stale[bot]