wled/WLED #2357: Web UI blue text on black background cannot be read (accessibility)

ptr727, 2021-11-24:
### What happened?

I cannot read the blue text printed on the black background.
I have to mouse select the text to be able to read it.

I tested with 12.0 and 13 Beta 5 (0.13.0-b5)
I use a 4K color calibrated monitor and Windows 11.

### To Reproduce Bug

Windows 11 (Pro, 21H2, 22000.348, 1000.22000.348.0)
Google Chrome (Version 96.0.4664.45 (Official Build) (64-bit))
WLED 0.13.0-b5
QuinLED-Dig-Uno V3
Navigate to "Security & Updates" page.
Cannot read blue text on black background.
Select text with mouse, text turns white, can read text.

There are multiple pages with blue text, this is just an example page:
[image]
[image]

### Expected Behavior

Expect to be able to read all the text.
Use a color pallet that can be read by most people, e.g. avoid dark on dark, light on light.
There are many references for accessibility and web design, e.g.:
https://designsystem.digital.gov/design-tokens/color/overview/
https://material.io/design/color/text-legibility.html#legibility-standards
https://www.w3.org/WAI/GL/low-vision-a11y-tf/wiki/Text_Colors
http://web-accessibility.carnegiemuseums.org/design/color/
etc.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b5 (build 2111170)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-11-24:
Is this more legible to you? (I cannot judge since original color is quite legible to me)
[image]

ptr727, 2021-11-24:
I can read that, thx.

Aircoookie, 2021-11-24:
100% agree, while I could personally read the links, the contrast is way too low. The purple (visited) links were near unreadable.
I have set the color to `#58a6ff` in master now https://github.com/Aircoookie/WLED/commit/37bd525638b5926e175409265a8b0b3a19993402
This is the color the GitHub dark theme uses for links.

blazoncek, 2021-11-24:
I opted for much shorter #0af 😉

ptr727, 2021-11-24:
Thank you for making the change.

Just as fyi, **any** blue colors are heavily deattenuated by "**night light**" (win) / "**night shift**" (mac) where blue is filtered at night time. E.g. look at the blue text on black with night shift on and off, big difference in visibility due to lack of contrast at night.

As an observation, one will rarely find dark blue on dark backgrounds used in accessible sites, e.g. https://material.io/design/color/dark-theme.html#anatomy

blazoncek, 2021-11-25:
Yet, the blue color is the default for links on most modern browsers...

ptr727, 2021-11-25:
Well, yes, but never dark blue on black, the default (on accessibility friendly sites, like github) blue is typically light blue, when night shift is on it reduces the blue, and as long as there are non-blue color components, it effectively lightens and increases the contrast, with just blue, it becomes more black.

It is simple to test and observe, change the default github theme to dark, refresh this page, then enable and disable night shift.
Anyway, the new blue works fine.

blazoncek, 2021-12-18:
Fixed recently. Closing.

closed 2021-12-18 by blazoncek