wled/WLED #2358: Odd and Even LED segments do not report the correct number of LED's

ptr727, 2021-11-24:
### What happened?

I'm retrofitting [Govee Patio](https://us.govee.com/products/govee-rgbww-wi-fi-bluetooth-outdoor-bulb-string-lights?variant=40861368778937) lights to use a QuinLED-Dig-Uno v3 and WLED 0.13b5.

The Govee lights are 2 x WS2811, 2 x color LED's, and 2 x white, **per** LED bulb.
[image]

I asked on [Reddit](https://www.reddit.com/r/WLED/comments/r0ykvp/using_wled_with_govee_rgbw_patio_lights/), and found a related [post](https://www.reddit.com/r/WLED/comments/ow57d7/govee_lights_mod_for_wled_ws2811/) suggesting the use of odd and even segments, odd for color, and even for white.

For my 30 bulb string, I configured 60 x WS281x LED's.
The first WS2811 controls RGB, and the second WS2811 controls White.
So I need to create a separate configuration for RGB and White control, each using its own WS2811.

Per the [wiki](https://github.com/Aircoookie/WLED/wiki/Segments):
```
Segment 0 has a Start LED of 0 and a Stop LED equal to the LED Count you defined in Configuration, LED Preferences. The Stop LED is not included in the Segment.
```
So I need to use 0-based numbering, and the stop position is not included.
I created two partitions (using 0-based numbering), one for odd WS2811's (0, 2, 4, ..., 58) controlling RGB, and one for even WS2811's (1, 3, 5, ..., 59) controlling White.

First partition: `start:0, stop:59, spacing: 1`, and reports "59 LEDs (30 virtual)".
Second partition: `start:1, stop:60, spacing: 1`, and reports "59 LEDs (30 virtual)".

The **59 LEDs (30 virtual)** is incorrect using 0-based and exclusion logic, it should be **60 LEDs (30 virtual)**.

### To Reproduce Bug

Configure 60 x WS281x LED's total.
Create two partitions, one for odd, and one for even.
Partition-0: `start:0, stop:59, spacing: 1`, save, and reports "59 LEDs (30 virtual)".
Partiton-1: `start:1, stop:60, spacing: 1`, save, and reports "59 LEDs (30 virtual)"

### Expected Behavior

"60 LEDs (30 virtual)"

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b5 (build 2111170)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-11-25:
You should be using ledmap.json and create simpler segments without spacing and such.

You can also use UI settings to switch between stop or length fields for segments if that helps.

ptr727, 2021-11-25:
Interesting, I'll explore the JSON route, seems more deterministic.
Other than the limited [wiki](https://github.com/Aircoookie/WLED/wiki/Mapping) info, is there info on more elaborate config, e.g. can I configure the WS2811's that drive the white LED's as white and not color?

As for this bug, seems it is an off by one counting error, maybe only affecting the display.

blazoncek, 2021-11-26:
Wiki is oudated use [kno.wled.ge](https://kno.wled.ge)

ptr727, 2021-11-26:
Looks like pretty much the same content, no explanation of detailed config?
https://kno.wled.ge/advanced/mapping/
https://github.com/Aircoookie/WLED/wiki/Mapping

blazoncek, 2021-11-26:
Wiki is moving to kno.wled.ge.
If there are updates they will not appear in wiki.
All of the kno.wled.ge is a shared effort and you can contribute too when you figure out a thing not mentioned there.

ptr727, 2021-11-27:
This worked, I used VSCode MinifiAll extension to convert pretty JSON to minified JSON:
```
{"map":[ 0, 2, 4, 6, 8, 10, 12, 14, 16, 18, 20, 22, 24, 26, 28, 30, 32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58, 1, 3, 5, 7, 9, 11, 13, 15, 17, 19, 21, 23, 25, 27, 29, 31, 33, 35, 37, 39, 41, 43, 45, 47, 49, 51, 53, 55, 57, 59]}
```

Thank you for the simplified setup, but the bug report is still valid, `59 LEDs (30 virtual)` should be reported as `60 LEDs (30 virtual)`.

blazoncek, 2021-12-13:
I have finally managed to test your original post and have concluded that WLED is working as expected and segment setup is correct.
When you configure spacing, you are virtually _stretching_ your LED strip by skipping LEDs (effectively reducing its length in pixels). The information about actual LEDs used is given in the _virtual_ remark (in parentheses).

Not a bug but misinterpretation on user's part.

closed 2021-12-13 by blazoncek

ptr727, 2021-12-13:
Can you please explain how 59 vs. 60 is a misinterpretation on my part (vs. a bug) when there are 60 physical LED's and 30 in the virtual set, not 59?

blazoncek, 2021-12-13:
A simple math will do:
59 - 0 = 59
60 - 1 = 59
There are 59 physical LEDs in each segment with spacing 1 (making every other LED not used in the segment, hence 59/2 = 30 virtual LEDs (rounded up)).
The segments created are overlapping (possible, but normally not recommended) with interleaving LEDs, making them appear in the desired configuration.

ptr727, 2021-12-13:
There are 60 physical LED's, in 2 groups (odds and evens), 30 in each pool.
0-based counting is 0-59 = 60.
1-based counting is 1-60 = 60.
There is no rounding, this is not an approximation of near 30 or near 60.
I think the display math is making an off-by-one error.

SerpentDrago, 2024-09-06:
> This worked, I used VSCode MinifiAll extension to convert pretty JSON to minified JSON:
>
> ```
> {"map":[ 0, 2, 4, 6, 8, 10, 12, 14, 16, 18, 20, 22, 24, 26, 28, 30, 32, 34, 36, 38, 40, 42, 44, 46, 48, [...]

I'm about to attempt the same thing you did with the same exact string of lights . 

so use this JSON , and then make 2 segments ??   30 & 30 ? 

Am i right that this mapping file will remap the even odd white / color into just one long 60 led virtual string . and then just use wled's segmentation to split it up ?

ptr727, 2024-09-06:
Yes, and sorry I no longer have the JSON.

softhack007, 2024-09-06:
> 0-based counting is 0-59 = 60.
> 1-based counting is 1-60 = 60.

Your math ik OK, however in WLED we handle start and stop differenty:
Start = index of first LED (base 0)
Stop = stop before this LED = start + length

So two segments of 30 leds means:
* seg1: start: 0  stop:30 (30 leds)
* seg2: start:30 stop 60

https://kno.wled.ge/features/segments/

> Segment 0 has a Start LED of 0 and a Stop LED equal to the LED Count you defined in Configuration, LED Preferences. The Stop LED is not included in the Segment. Currently you can create a maximum of 10 segments. Presets 1-15 use only Segment 0 by default. Preset 16 is the only Preset that saves settings for Segments 1-10.

softhack007, 2024-09-06:
> First partition: `start:0, stop:59, spacing: 1`, and reports "59 LEDs (30 virtual)". 
> Second partition: `start:1, stop:60, spacing: 1`, and reports "59 LEDs (30 virtual)".
> 
> The **59 LEDs (30 virtual)** is incorrect using 0-based and exclusion logic, it should be **60 LEDs (30 virtual)**.

Please try this
* seg1: start 0 stop *60* spacing 1 (60 leds, 30 virtual)
* seg2: start 1 stop *60* spacing 1 (59 leds, 30 virtual)