wled/WLED #2359: Esp8266 and ws2801 ant binary doesnt work for my card

insproad, 2021-11-25:
I use two esp01 without any problem but when I try with my new card. Any binary doesnt show Wlep-Ap, have you any fix bin file for me, I want to use WS2801 leds
[image]
.

blazoncek, 2021-11-26:
Use 4k7 pull-up on data pins. Use GPIO2 & GPIO1 for data & clock.

insproad, 2021-11-28:
but problem not about the connection. wled ap isn' t working

blazoncek, 2021-11-28:
Do you want help or not? Listen to what I say.

insproad, 2021-11-29:
> Do you want help or not? Listen to what I say.

I will try that, but I want to learn. When we dont connect the leds, can we see wled ap? If we can see, that isnt make sense.

austwhite, 2021-11-29:
Try the pull-ups as requested.  Some boards need certain pins at a the correct state to start the board.  If it doesn't work, you might have a faulty board or one that has weird compatibility issues.   
Also, try to remember none of us are paid to do this.  We try to help best we can based on what information we have, we are not able to hold hands for every scenario that comes up

mouzg, 2021-12-05:
I have the same board and the same problems, soldered the pull-ups on tx rx as suggested but made no difference. On the other hand WLED 8.5 is working flawlessly so I'm guessing it must be something similar to [this](https://github.com/Aircoookie/WLED/issues/946)

blazoncek, 2021-12-05:
GPIO2 and GPIO1 and GPIO0 need pull-up resistors, not GPIO3
I have found out that some ESP01S (perhaps also ESP01) boards have a very strong on-board LED resistor making it difficult to use GPIO1 for anything but button.
Try GPIO2 and GPIO3 for WS2801 LED output.

makeroutlet, 2021-12-05:
i wouldnt say you need pull resistors.. im running ws2801 on esp8266 directly no mods and works fine

closed 2022-02-04 by blazoncek