wled/WLED #2361: Preset for 0 dim and all black (effectivly off) doesn't hold

JPElectron, 2021-11-26:
### What happened?

After loading a preset that is effectively "off", both units when sync'd will turn back on after a few seconds and resume a previous preset which is a playlist

### To Reproduce Bug

On the "main" unit where Sync is toggled on...
Create preset ID 1 with Dim value = 0 and solid color black (effectively off)
Create preset ID 2 that is a playlist of preset ID's 3 thru 7 (each are just loading built-in effects) that switch at 20 sec intervals
Using the webpage, or HTTP API, load preset ID 2 - works, playlist is running
Using HTTP API, load preset ID 1 - both units go off, for awhile, then go back to playing preset ID 2 - not sure why.
Using the webpage to load preset ID 1 - both units go off, and stay off - as expected.

### Expected Behavior

Expected behavior is they stay off, per preset ID 1, regardless if preset ID 1 is loaded by webpage, or by HTTP API

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b5 (build 2111170)

### Which microcontroller/board are you seeing the problem on?

Other

### Anything else?

Tested with a pair of Dig-Uno v3 boards, both to same AP via WiFi

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-11-26:
Instead of describing of what each preset is, please post the content of your `presets.json`.
Try to minimize it to the bare minimum to reproduce the bug.

blazoncek, 2021-12-18:
Any feedback would be appreciated or I'm closing.

closed 2022-04-17 by Aircoookie