wled/WLED #2362: Webpage shortcuts (for, Effects, Presets) and way to hide Config button

JPElectron, 2021-11-26:
When loading the IP of the module, it always starts at the "colors" page

Need URL shortcuts to go directly to Effects and Presets page for "older" or "kid" users who are "impatient"

Propose using URL "sub-folders" or "query-string" to go direct to those other pages...
http://[IP_Address]/Effects
http://[IP_Address/Presets
or
http://[IP_Address]/?load=effects
http://[IP_Address]/?load=presets

Also need way to hide the config button, useful when module IP is linked from other home automation pages...

http://[IP_Address]/Effects/no-config
or
http://[IP_Address]/?load=effects&config=false

...this could simply set a session state of config=false so no matter which pages are hit afterwards the config menu button stays not visible for the duration of that browser session)

blazoncek, 2021-11-26:
For that purpose I have developed a _simplified_ UI available in my fork as an experimental feature.
I would be grateful if you can test it.

softhack007, 2023-10-07:
@blazoncek simplified UI is part of official WLED now, should we close this request as implemented?

blazoncek, 2023-10-07:
I do not think we should close it as opening particular tab is not implemented (useful on mobile).

blazoncek, 2024-04-30:
@w00000dy could you look into this issue if you can conjour something up? 😉

commit d2984e9e (2024-05-03): add Webpage shortcuts, resolves #2362
closed 2024-10-27 by blazoncek