wled/WLED #2364: Enhancement: Per port current limiter

larrywal, 2021-11-28:
Problem: When I’m running multiple strings of props off a dig quad, some off data1, some off data2, etc… I want to make sure none of the prop strings blows one of my 5A fuses.  But I can’t set a 5A current limit because that applies to the whole board.

**Describe the solution you'd like**
Let me set a brightness limit per data port AND one for the whole board.  The whole board obviously would trump whatever is specified on a data port.

**Describe alternatives you've considered**
I can’t think of another solution.  I can limit brightness in xlights, but that won’t help for WLED only scenarios.

henrygab, 2022-05-02:
I would recommend linking the power limits to be **_per-segment_**, rather than per-GPIO.

For complex setups with a single output, the segments could correspond to the groups of pixels drawing power from a given power source.  (e.g., first 250 pixels on 20A power supply, next 250 pixels on a second 20A power supply, and final 50 pixels on a 5A power supply).

For those using separate power supplies per GPIO, one segment could correspond to each GPIO.

Open questions:
* What if you need to share power across multiple segments?
* What if you need to share power across two GPIO (and are using 3x+ GPIO)?
* What policy to expect if not all segments require the same reduction?
    * Should the policy be that all segments are power-limited by the greatest amount?
    * Should the policy be that each segment is power-limited independently?
    * Should it be possible to group segments, and apply the power-limit policy to groups of segments?
* How can the desired solution be implemented with reasonable RAM and processing overhead?

blazoncek, 2022-05-03:
There are multiple layers to this problem.
First, what are your power supplies and what are your power demands. If you use multiple power supplies how are injection points managed, do they span segments or not, etc.
Second, segments can (and often will) change, how would you manage that? Taking into account above.
Third, limiting power consumption per strip (AKA output) may seem reasonable but may not take into account that a single strip may be powered by multiple (different) power supplies (not very common but used in large installs) while maintaining a singe segment.

And more. TL:DR

Aircoookie, 2022-05-03:
I also think a per-segment limit would be too complex.
A per port limiter has more uses, primarily if you mix different kinds of LEDs, e.g. digital and analog, in one project.

spaethdiy, 2022-07-04:
I will have such the same problem with my planned setup.
At the moment I plan to control 1440 WS2812 LEDs with one to two GPIOs. Nevertheless I need three power supplies which are not supporting parallel connections. 
So I think it would be great if it would be possible to configure which LEDs are connected to which power supply and the maximum rating of each supply individually. 
I think per GPIO or per segment is not flexible enough.
If I could wish for something I would prefer a configuration part on the "LED Preferences" page similar to the LED to pin configuration where every LED section can be linked to a separate power supply with its amp rating.

At the end a big praise for the already great software.

kgreene1983, 2023-06-28:
> I also think a per-segment limit would be too complex. A per port limiter has more uses, primarily if you mix different kinds of LEDs, e.g. digital and analog, in one project.

I just posted a request for this on the discourse forum.  

It would be great to have it per output limit.. Would also be handy if there was master brightness that could be scheduled..

I Use WLED for Christmas Lights. In New Zealand, Christmas is mid-summer and it doesn't get dark until 9pm, so have to run at full 6pm -9pm and then drop them down so the neighbors don't go made.

blazoncek, 2023-06-28:
For brightness management, just save presets with brightness excluded, then schedule brightness only presets when desired.
Brightness only presets can be entered using JSON API.

kgreene1983, 2023-06-28:
I'm using xlights and scan to control my wled so presets isn't an opinion

blazoncek, 2023-06-28:
> I'm using xlights and scan to control my wled so presets isn't an opinion

Then manage all aspects in xLights.
Read my post above.

blazoncek, 2023-11-16:
8f7f9ec3676af6d1da8a9b5114a43d53dbcc0fd7

blazoncek, 2024-03-22:
Available in 0.15.0 beta 1.

closed 2024-03-22 by blazoncek