wled/WLED #2366: Save light config as preset not possible from preset

snoofx, 2021-11-28:
### What happened?

When trying to save a new preset via a different preset with the JSON command `psave` or the HTTP command `ps` both result in weird behavior. When saving a preset with the following JSON command : 
```json
{"psave":1}
```
The command is just replaced with empty brackets `{}`.

When using the HTTP command:
```
PS=1
``` 
The preset is saved successfully but when loading the preset the content of the new preset is just the API command instead of the current light status.

### To Reproduce Bug

Add a preset with one of the commands described above. The preset command will either be replaced on save or when loading the preset the newly created preset will just be a duplicate of the other preset. 

### Expected Behavior

- JSON: Adding the `psave` value pair should be possible. 
- HTTP: When executing `ps` from a preset the current state should be saved, not the `ps` command

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0-b5 (build 2111170)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I am trying to use this to save and later restore a state after a GPI event.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-11-28:
If you use JSON you have to include all preset items in the JSON API.

snoofx, 2021-11-28:
I'm sorry I am not sure what preset items you are referring to. Even when adding more parameters to the JSON object the `psave` key is always removed on save.
When sending the same commands using curl, both are working as expected.

blazoncek, 2021-11-28:
"psave" is just an instruction to save the rest as a preset, some other keys also get removed.

snoofx, 2021-11-28:
The documentation for `psave` says "Save current light config to specified preset slot."

When executing the following command:
```curl -X POST "http://10.0.20.20/json/state" -d '{"psave":1}' -H "Content-Type: application/json"```

The currently active light config is stored into the preset, just as expected. 

But when using the same JSON snippet in a preset, the issue arises.

Your response suggests that the JSON object is invalid, but if the object works when calling the API directly shouldn't the behavior be the same within a preset?

blazoncek, 2021-11-29:
When you are trying to save a preset using JSON API the whole JSON request has to be structured as a preset, not just `{"psave":1}`.
You will need to define every parameter of preset if you want something to be saved.
On the contrary HTTP API uses current WLED state to save it as a preset.

closed 2021-11-29 by blazoncek