wled/WLED #2368: Cannot Adjust Number Of LEDs on 0.13.0-b5

MonsterMuffin, 2021-11-29:
I have a number of WLED instances running on Wemos D1 Minis. These were all running WLED_0.12.1-b1 and I recently went to add some more, so I updated my previous installs.

I have noticed I am no longer to change the number of LEDs in the settings page as can be seen here:

[image]

I am having to flash WLED_0.12.1-b1, change this setting, then update to WLED_0.13.0-b5.

Is this expected behaviour? 

Thanks.

fishbone-git, 2021-11-29:
it is intended.  see #2354

MonsterMuffin, 2021-11-29:
> it is intended. see #2354

My apologies, I see this now. I was being stupid.

Thank you very much!

closed 2021-11-29 by MonsterMuffin

Bohan-Song, 2022-06-10:
> > it is intended. see #2354
> 
> My apologies, I see this now. I was being stupid.
> 
> Thank you very much!

Sorry my English is not very good, I have the same problem led numbers can only be set to 30, why? Will the number be calculated automatically here without manual setting?

blazoncek, 2022-06-10:
You set number of LEDs on each GPIO separately.

Bohan-Song, 2022-06-10:
> You set number of LEDs on each GPIO separately.

What's mean, I can't understand...

blazoncek, 2022-06-10:
Scroll down...