wled/WLED #2369: APA106 flickering, timingproblem

Bubbleman75, 2021-11-29:
Hello

I have a problem with APA106 LEDs, they are flickering and show random colours. I have 17 of these LEDs in a row. The length is 50cm. All cables are short and neatly soldered. I have tried everything. With and without level shifter. With and without resistor and capacitor. The power supply has 5 Amper. I have tried an ESP8266 and an ESP32. It does not get better.
I think it is the timing. Although it says everywhere that APA106 = WS2812B. But this is not true! The timing is different.
I would be very happy if you could include this LED type in WLED. Otherwise I would also try to change the timing myself, if I knew where that is possible.

Thanks a lot
Thomas

[image]

blazoncek, 2021-11-29:
Look at NeoPixelBus.

Bubbleman75, 2021-11-29:
This is certainly a powerful library. But I want to use WLED because it is ready to go. I use WLED without any Problems with a 500 LEDs SK6812 RGBW Strip!

blazoncek, 2021-11-29:
NeoPixelBus is the foundation WLED is built upon.

dendel71, 2021-11-30:
Hi @Bubbleman75,

I have the same issue. All effects can't stop flickering. Even "Solid".
LEDs = WS8211
Length = 469
Board = ESP8266
Firmware version =  0.13.0-b4

Any tips how to solve or which configuration to play with?

blazoncek, 2021-11-30:
Use level shifter and make sure all connections (and GND especially) are good.

dendel71, 2021-12-06:
I used a level shifter, ....it did not work. Then I used a scacreficial LED...it did not work. BUT, then I replaced the cable between MCU and LED string. That did work!
Old cable was probably too thick and too much resistance. Before MCU I used an Arduino and that did not to have an issue.

Airrr17, 2021-12-24:
Same here. My setup is 53 leds in a string. First 30-40 works somehow close to desired result, with flickering and red-green colors swapped. Near the end of string there is only random colors.
Sure this is problem with timings.
Will you add support for apa106 in any future?
Thanks.

blazoncek, 2021-12-24:
What level shifters do you use?

PureTek-Innovations, 2023-01-08:
I have some APA106s that I'd like to use with WLED.
I have posted also here:
https://wled.discourse.group/t/apa106-support/7037

itstefanjanos, 2023-12-11:
I've made a pull request for APA106 support and I've just tested with Wemos D1 mini and PL9823 (It has same timing then APA106).
https://github.com/Aircoookie/WLED/pull/3580

blazoncek, 2024-03-24:
Implemented in #3580 
available since 0.15.0-b1

closed 2024-03-24 by blazoncek