wled/WLED #2371: Failed VS compilation platformio.ini

pjhatch, 2021-11-30:
Created a fresh clone and followed all instructions installation/compilation and updating paltformio.ini.

Compilation failed while configuring d1_mini

d1_mini        FAILED    00:00:01.045

Is there some other configuration I need to set up? What am I missing?

In some of the output I see:
Processing d1_mini (board: d1_mini; platform: espressif8266@2.6.2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.2) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.7 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Library Manager: Installing git+https://github.com/lorol/LITTLEFS.git
UserSideException: Please install Git client from https://git-scm.com/downloads:
  etc.

blazoncek, 2021-11-30:
It says in plain English: **Please install Git client**
And it even gives instructions where from.

pjhatch, 2021-11-30:
Awesome! This solved my problem. Might want to add this step in documentation - although it seems obvious.

closed 2021-11-30 by blazoncek

Aircoookie, 2021-11-30:
Added to documentation :)