wled/WLED #2372: Circadian light automation to change light based on time of day

magiva, 2021-12-01:
**Is your feature request related to a problem? Please describe.**
no problem, feature to have circadian lighting mode like what home assistant does with 
https://community.home-assistant.io/t/circadian-lighting-custom-component/61246

**Describe the solution you'd like**
A mode to have the lights change to the white/yellow colour based on time of day - circadian

**Describe alternatives you've considered**
connecting home assistant and using the circadian lighting component but i want this stand-alone

**Additional context**
Would be nice to extend this feature to have specfic colours faded in at scheduled times, maybe we like blue in the morning and red at night for example.

Thank you for your ideas for making WLED better!

blazoncek, 2021-12-18:
You can create presets and playlists to approximate your desired behaviour.
You can trigger those playlists at certain time using Time & macro config page.

magiva, 2021-12-18:
pretty much anything is possible that way,  i was looking more towards
automation,  so that would adjust as the year does.

blazoncek, 2021-12-18:
Too much hassle on a little device.
Can be solved with external automation as noted in OP.

closed 2021-12-18 by blazoncek