wled/WLED #2376: IR recv return random values

dDaYb, 2021-12-02:
### What happened?

Same IR remote button press echos different code on the serial monitor. There is no such problem in 0.12.0 vesion

### To Reproduce Bug

tsop 4838 connected to pin d2 on nodemcu v3, then press on IR remote same button with patience.

### Expected Behavior

receive same code

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2021-12-02:
Hi, thanks for the report. Are you using any LED pin other than GPIO2/D4?

dDaYb, 2021-12-02:
Yes, I use D6 and D4 for 2 ws2812 strips. And D0 and D3 for 2 momentary buttons.

Aircoookie, 2021-12-02:
Ok, D6 (GPIO12) is a problem, it uses software driving, which will interfere with IR receiving. If you change that strip to use pin TX (GPIO1), IR should work :)

dDaYb, 2021-12-02:
ok, but now I get garbage in console instead of a IR codes? And why its ok with 0.12 version, but not 0.13.0-b5?

blazoncek, 2021-12-02:
GPIO1 is Serial output from ESP.

dDaYb, 2021-12-02:
I guessed)) which pin should I use instead?

blazoncek, 2021-12-02:
For LED output or IR?
LED: 1, 2, 3
IR: 12,13,14

sn0wst0rm, 2021-12-07:
Hello, I have the same issue. I have a WS2811 strip connected to GPIO5, and IR receiver to GPIO12. Every time I press the same button I get a random code. The only right code I get is when I long press a button, getting 0xFFFFFFFF.

How can I solve this problem?
Thanks

dDaYb, 2021-12-07:
> Do as suggested: use GPIO 1, 2 or 3 for LEDs

Every time I press the same button I get a random code even if I connect
only ir diode without any leds.

closed 2022-04-27 by stale[bot]