wled/WLED #2380: Long reaction time with IR Sensor KY-022

AWSb24, 2021-12-03:
### What happened?

By pressing any button on my 24 key remote several times until the ESP8266 changes color or switches on or off.

### To Reproduce Bug

I connected the IR sensor to GPIO4, 5, and 12. There is no change in behavior.

### Expected Behavior

Immediate reaction

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-b5

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I comparing ESPHome with a self-writing config YAML with WLED.
I also flashed ESPHome on the same device to check whether it is an electronic or software issue.
On ESPHome it works flawlessly.

Yaml segment for on button in ESPHome for NEC remote command:
```
  - platform: remote_receiver
    id: "onbutton"
    nec:
      address: 0x00F7
      command: 0xC03F
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-03:
Which pins are used for LEDs?

AWSb24, 2021-12-03:
I have no LED strip connected yet but I left it on default (GPIO2).

blazoncek, 2021-12-03:
Can you see IR codes displayed on serial output when you press the button?

AWSb24, 2021-12-03:
This is what I get if I press the button several times or do a long-press the last 5 or 6 commands.

`IR recv
0xCE1972FD

IR recv
0xF7C03F

IR recv
0xCE1972FD

IR recv
0xFFFFFFFF

IR recv
0xCE1972FD

IR recv
0xCE1972FD

IR recv
0xCE1972FD

IR recv
0xCE1972FD

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xCE1972FD

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xFFFFFFFF

IR recv
0xCE1972FD
`

blazoncek, 2021-12-03:
0xFFFFFFFF is sign for repeating last button (press&hold aka longpress)
What is the time between outputs?

AWSb24, 2021-12-03:
I get approx. 7 commands per second. I wonder why I get `0xCE1972FD` and not `0xF7C03F` ?
It seems the signal is not inverted in the ESP.

blazoncek, 2021-12-03:
that would be expected time.