wled/WLED #2381: First Time Adding Usermods (Sensors_to_MQTT) Casuses Crashes (Lots of info inside, please help)

DudeShift, 2021-12-03:
### What I am trying to do:
First time trying to add a usermod to WLED. After editing Sensors_to_MQTT, compiling, flashing, the usermod is _mostly_ working (more on that later) HOWEVER after 1~5 minutes it crashes WLED and reboots again and again. I want to use Sensors_to_MQTT for the Home Assistant device setup/discovery.

### Prereqs before continuing:
- I'm new to platform IO so please go easy on me. I know I probably made a error along the way and need help. First time compiling source code from someone else's project.
- Due to (I assume) outdated usermod information, I am editing files directly (platformio.ini and usermod header files). I understand that there is a override and my_config files in the source but without any updated guide to follow, I am following the steps stated in the Sensors_to_MQTT usermod.
- I am flashing a D1 Mini ESP8266 with ESPHome-Flasher 1.4.0. 
- I am using WLED 0.12.0 (build 2104020) to compile. 
- I am sleep deprived so I hope this is enough info for someone to make a guess on what I did wrong.

### Changes made to Sensors_to_MQTT and platformio.ini
For Sensors_to_MQTT I:
- Removed anything relating to BMP280 and CCS811 sensors. 
- Switched anything relating to Si7021 to HTU21DF sensor. The only _meaningful_ difference I can find looking at Adafruit libraries is what the response from the sensor should be over I2C during the `[sensorname].begin()` function. I am using the HTU21DF and that's why I changes libraries.
- Added a conversion from celsius to fahrenheit.
- I made all my edits in the original Sensor_to_MQTT file before moving it to the wled00 folder.

For platformio.ini I:
- Added the semicolon and extra space to comment out the release binary and took out the semicolon and extra space to uncomment out the `default_envs = d1_mini`.
- Uncommented the build flag `-D USERMOD_SENSORSTOMQTT`
- Added `adafruit/Adafruit Unified Sensor @ ^1.1.4` and `adafruit/Adafruit HTU21DF Library @ ^1.0.5` to the `lib_deps`.

### Compile and Flash
Following the guides on [https://kno.wled.ge/basics/](https://kno.wled.ge/basics/) I compiled using platform IO. _RAM: 49.4% (used 40472 bytes from 81920 bytes)_, _Flash: 67.7% (used 707528 bytes from 1044464 bytes)_ so I am not out of memory I think from adding a usermod that I trimmed down. I then flashed using ESPHome-Flasher. During the flash it does say that it erase the flash first.

### Setting up WLED and MQTT
I am able to setup WIFI on WLED and enter my Home Assistant MQTT broker info. I make sure to select enable MQTT and power cycle the WLED D1 Mini. I can see within Home Assistant MQTT broker that I am getting the device MQTT setup from the Sensor_to_MQTT usermod and the Temp + Humidity data/numbers from my HTU21DF sensor (there is one issue but I will bring that up later).

### Crashes and Reboots
Here is where I can see that the D1 Mini is crashing. Using Termite (Serial Monitor) after about one minute to five, there is a wall of text being sent through Serial related to json, WLED color palettes, and general garbage. [Termite Wall of Text.txt](https://github.com/Aircoookie/WLED/files/7647760/Termite.Wall.of.Text.txt) Can confirm crashing from also turning off the light power from WLED webpage, wait for a crash, and on reboot the light power will be back on (since I think be default a power cycle will always enable the light power)

### EXCEPTION DECODER
Included with the wall of text Serial output is the EXCEPTION DECODER information (included with Termite output). [I found this guide that breaks down the important parts of the Exception Decoder.](https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html)

- `Soft WDT reset` at the top
- Gives a stack (will come back to this) 
- `rst cause:2`, which according [ESP8266 Reset Causes PDF](https://www.espressif.com/sites/default/files/documentation/esp8266_reset_causes_and_common_fatal_exception_causes_en.pdf) on page 4 says "External reset or wake-up from Deep-sleep" but also that rst cause won't change on a Software WDT reset so it could be from a time when I pressed the reset button on the D1 Mini.
- `Fatal exception (%d)` was mixed in with the garbage wall of text. Can't find any useful info since [usually in the Exception Decoder there should be a exception line with a number from 0 to 64 according to same guide different page.](https://arduino-esp8266.readthedocs.io/en/latest/exception_causes.html)

Using the stack and installing the EspExceptionDecoder tool via ArdunioIDE, opening up the firmware.elf (how festive) produced by platformio, copy pasting the stack in I get decoded stack results of:
`0x402550e4: vsnprintf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/vsnprintf.c line 42`
and that's it. It isn't much to go on and that's why I turned for help.

### Things that I have tried
- Changing from ESPHome-Flasher 1.3.0 to 1.4.0
- Changing WLED version from latest WLED beta release 0.13.0-b5 to WLED 0.12 (what I had before on my D1 Mini, and two different ESP8266s that I not trying to modify)
- Trying to research what the vsnprintf error is. (I have some coding skills but not enough to go digging into stdio libraries / stack tracing problems.)
- Adding some personal printf debug lines to make sure code is executing. 
- Looked on WLED github issue page and discourse page for similar issues.

### Possible Causes for my Issue
- In Home Assistant, when Sensor_to_MQTT sends the config for the MQTT device (in `mqttInitialize()` / `createMqttSensor()`), The device name for humidity sensor works fine but the device name for temperature sensor is **null**. It gets **null** from a call to get the variable `mqttClientID` in `createMqttSensor()` where it gets the correct `mqttClientID` three other times before in the same `createMqttSensor()` function. The call to make this MqttSensor for humidity works fine but not for temperature which happens right before. This is a still problem I need help with even if my D1 Mini wasn't crashing and a return of **null** is never good HOWEVER not sure if this is the cause of the crashing. 
- In the garbage wall of text there is references of json. Although in Sensor_to_MQTT uses a json document, I think these references of json are for WLED since there are like WLED settings near? See the Termite wall of text attached to see what I mean. [Termite Wall of Text.txt](https://github.com/Aircoookie/WLED/files/7647760/Termite.Wall.of.Text.txt) I don't think its WLED fault here, and not sure why this is outputted on Serial before each crash. For a while I thought it was a out of memory problem since there is so much garbage being put on Serial but not sure if that is the real issue.
- In Sensor_to_MQTT, at the end of the `updateSensorData()` function there is a `printf` to print the new sensor data to Serial. Although I can confirm that this function is executing since I am getting new / different sensor data being sent over MQTT, only sometimes I have seen this print execute BUT the floats of the sensor data are never show in the Termite Serial Monitor (just "f, humidity" no actual numbers but new data is being sent over MQTT. Is it possible that something **THIS** is what the `vsnprintf exception` is referring to? Feels like I am pitching a long shot but I am completely out of ideas and energy.

### Please help me
Please help me. I hope I gave enough information for someone to make a guess on what to do next. Adding usermods in theory shouldn't be this hard but here I am. 

### Files:
Since I can't attach code files here, I uploaded them to a pastebin as that was the quickest way to share them and for me to get to sleep.
- [Edited Sensor_to_MQTT Usermod](https://pastebin.com/GpfzUGQu)
- [Edited PlatformIO.ini](https://pastebin.com/rpexriqh)
- [Termite Wall of Text.txt](https://github.com/Aircoookie/WLED/files/7647760/Termite.Wall.of.Text.txt)
- Home Assistant Device **Null**:
[image]

blazoncek, 2021-12-03:
TL:DR; Contact original usermod creator and always use latest version if possible.

DudeShift, 2021-12-03:
Thank you for responding to my issue.

> Contact original usermod creator

I can do that. I wrongly assumed that since the usermod was v2 and included with the source of WLED that it didn't have problems. I also wanted to asked for help from a large group of people then just the creator (two (in this case more) heads are better then one).

> always use latest version if possible

I did try using the latest WLED beta release 0.13.0-b5 first when I downloaded the source. However when I ran into the crashing issue, I went back to the last stable / full release WLED 0.12 to hopefully rule out any possible beta issues. The last version I had installed on three different esp8266 with no issues was WLED 0.12 so that's why I stuck with the last stable/full version for trying to fix the issue.

blazoncek, 2021-12-03:
Any updates will be based on 0.13-b5. So you better use that.
I will also urge you to join Discord for any support issues.

DudeShift, 2021-12-03:
Okay I will go back with 0.13-b5 and make a post on discord after trying to few more "maybe this is the problem fingers crossed" fixes.

blazoncek, 2021-12-03:
Be sure to use debug build and monitor serial output.

DudeShift, 2021-12-03:
So for anyone that is reading this in the future, I was able to solve what the issue was. But let me share what I did leading up to that.

### Re-Setting Up:

I went back to compiling for the latest WLED 0.13-b5 release.

I went and remade all my edits to Sensor_to_MQTT. I did realize that I did change the original if statement for checking if MQTT is connected to a version that was in the Temperature usermod when troubleshooting MQTT. I don't think this was the issue.

I correctly used the my_config file to define what usermod I was using. I still edited the original platformio.ini file.

### Trying d1 mini debug:

I tried using the d1 mini debug build like blazoncek recommend but it kept crashing after initializing sensors (once it exits the setup function and goes into the main loop) with `rst cause:4`, which [Hardware wdt is the last resort of ESP to tell you that application is locked up](https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html). 

I went back and compiled using normal d1 mini and then compiled for the debug build so see if I needed to compile for the normal version first, but debug build kept crashing with hardware wdt. 

As much as I wanted to use the debug version to compile, I never figured out what was the problem with that.

### Went back to normal d1 mini compile:

I went back to using normal d1 mini to compile and got the same software wdt crashes as before.

I then changed the printf statement of printing out the sensor data to serial in `updateSesnorData` to instead of printing floats using `%f`, I used `%4.2f` (4 min characters printed, with 2 digits of precision). This still DID NOT print the actual number into my Termite Serial Monitor **HOWEVER after a few minutes did not see a crash after a few minutes.** I decided to comment out / remove that printf statement. 

I also increased the json document size when creating a mqtt sensor in `createMqttSensor` from 300 to 350 and that fixed the device name for the temperature mqtt sensor in Home Assistant not to be **null**! I figured that since I was using a longer then default device name that it was too big to fit inside the json doc multiple times.

I have tested working for 30mins without a crash so I call that a win. 

### TL:DR for what was the issue:

- `updateSesnorData` printf statement might have been causing issues since after removing I didn't see crashes for a few minutes. Would make sense comparing to the vsnprintf stack error I was having
- json document for `createMqttSensor` needed to be increase to fit a custom/longer device name then default. This might have been an issue due to it returning null before.

closed 2021-12-03 by DudeShift

blazoncek, 2021-12-04:
By debug I was referring to include -D WLED_DEBUG during compile (and not debug_build).