wled/WLED #2382: Add api/macro-option to start effect at first LED

dekiesel, 2021-12-03:
**Is your feature request related to a problem? Please describe.**
I am trying to start an effect always at the same LED. I'd like to program a "Fade-In/Out" effect for when my kitchen lights are activated.

**Describe the solution you'd like**
A api/macro option to start an effect at the first LED (or the option to start it at the n-th LED).

**Describe alternatives you've considered**
I have used Presets to react to a button push. The preset correctly starts the Wipe-animation, but the animation starts at a "random" LED. I am guessing there is some loop running in the background that "decides" which LED is getting started.

**Additional context**
Other people have similar issues, like [here](https://wled.discourse.group/t/run-led-along-stairs-start-effekt-alwas-with-step-1/3977/2)

Thank you for your making WLED!

blazoncek, 2021-12-03:
This would depend on the effect used. Not all could use such feature without rewriting.

Aircoookie, 2021-12-03:
Hey, you can achieve this by adding `"tb":0` to your preset JSON, which will reset the animation to start at the first LED.
(Note this only works in 0.13 betas, not yet in 0.12)

softhack007, 2023-01-06:
Closing, as a workaround exists, and this issue has no activity any more since 12 month.

closed 2023-01-06 by softhack007