wled/WLED #2383: DDP receiving strip white channel not working

blark, 2021-12-03:
### What happened?

SK6812 White channel is not updating on the virtual strip (DDP network)

### To Reproduce Bug

Set up two strips with 60 pixels, one connected directly and controlled via GPIO16 on ESP32. The other is a virtual strip over DDP, (also using ESP32 GPIO16 on the receiving WLED). 

1. Set effect to solid, color to default.
2. Adjust white channel brightness slider

Observe that only the hardware controlled strip changes brightness of the white channel.

3. Adjust the colour temperature (add warm) and the white LEDS do light up on the virtual strip
4. Adjust the colour temperature to the far left (warm) and the white LEDs turn off on the virtual strip but not the hardware controlled strip.

### Expected Behavior

DDP network virtual strip should match the hardware controlled strip

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.0 b5

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-03:
DDP protocol (currently) does not specify RGBW support.
We requested an update of the specification to support RGBW (and RGBW+CCT) from the authors.

blark, 2021-12-03:
Ahh, thanks! Sorry for the false bug report then!

Aircoookie, 2021-12-06:
I just got word from the author of the DDP spec! Basically the current data type byte is just in an idea/concept state and not formally defined. xLights, sending 8 bit RGB data, should have a data type of `0x0A` based on the ideas given in the spec, but instead it uses `0x01` (and additionally seems to support some sync/discovery feature on `0x80`).
WLED too currently ignores the value of the type field assuming 8-bit RGB, and network DDP just sends out `0`.

Just replied to the mail encouraging the author to formalize the data types. If he prefers to keep it the current way though, there is still the "custom" range 128-255 which we could use to "privately" discern RGBW and RGBCCT types (e.g. `0xD3` = RGB, `0xD4` = RGBW, `0xD5` = RGBCCT) and treat all other values as if they were RGB.

Karshilistic, 2022-05-26:
@Aircoookie Any updates on this? Was surprised to find that it doesn't support RGBW. Is there a way we can contribute to this?

blazoncek, 2022-05-26:
> @Aircoookie Any updates on this? Was surprised to find that it doesn't support RGBW. Is there a way we can contribute to this?

No. Protocol still doesn't define RGBW. It does allow it but in a non-standard/not defined way.
[DDP](http://www.3waylabs.com/ddp/)

closed 2022-09-02 by Aircoookie