wled/WLED #2385: I have a question

MagdicsAron, 2021-12-04:
Hey👋
First you need to know that my english is very Bad.
I have that quastion that i have a esp8266 and i want to run a 12V 4pin RGB led strip but the esp runs on 3.3V i have a converter but i dont know the data wires can you guys/girls help

Aircoookie, 2021-12-04:
Hey! The data voltage is 5v even for a 12v strip. In many cases, connecting GPIO2 (D4) to the LED strip data in will still work, but it is highly recommended to use a 3.3v->5v level shifter for the data signal, like the sn74ahct125n :)
To _power_ the esp8266 from 12v, you'll nedd a DCDC converter to get 5v for most boards, or 3.3v for ESP01.

MagdicsAron, 2021-12-04:
Hey thank you it is helpful
I alredy have a DCDC 12v to 5v converter and i have bought a IRL520NPBF MOSFET Transistor coud that work?

MagdicsAron, 2021-12-04:
And can you Show me the channels on the esp becose i dont know (is it D1 D2...? ) thank you for the help by the way

blazoncek, 2021-12-04:
Please join Discord where a greater audience will be able to assist you.

sansillusion, 2021-12-04:
You will need a level shifter, and make sure all your grounds are connected
together.

MagdicsAron, 2021-12-04:
Yeah i know thank you @sansillusion but will the IRL520NPBF work

Aircoookie, 2021-12-04:
Ah now I understand, you have a PWM LED strip, not a digital one. The IRL520NPBF seems like it is likely to work. For PWM data pins, you can use almost any, e.g. 13,14,15 (D7,D5,D8)

Aircoookie, 2021-12-04:
Pins are configurable in WLED LED settings. Use "PWM RGB" LED type

MagdicsAron, 2021-12-05:
Thank you for your help

closed 2021-12-05 by Aircoookie