wled/WLED #2386: Drip effect in reverse on LEDs 2~3 crashes WLED/resets device

Shoopity, 2021-12-04:
### What happened?

When Choosing Drip as the effect in a segment that includes LED # 2 and/or 3 then putting the effect in reverse, WLED, or the device, resets.  Findings in the Reproduce section.  ESP32 Devkit Ver 4

### To Reproduce Bug

Defined Segment 0 as: 0~50
Do one of the following:
    Select Drip as the effect then reverse the segment
    Reverse the segment and select Drip as the effect

Experimenting:
It seems it only crashes if a segment includes LEDs 2 and/or 3.  Meaning I can define a segment 0~1, then another segment 4~50, reverse either or both with Drip as the effect, and everything appears hunky-dorey.  It's only when I have 0~2, or 3~50 (or 0~3, 2~50, etc.) that it crashes.  I _can_, however, make a segment 2~3 and it doesn't crash, but as soon as I change it to 1~3 or 2~4, it crashes.

### Expected Behavior

Dripping in reverse should work fine, regardless of segments, LEDs included, or direction.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.12.0 (build 2104020)

### Which microcontroller/board are you seeing the problem on?

ESP32, Other

### Relevant log/trace output

```shell
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dc751  PS      : 0x00060e30  A0      : 0x800dc843  A1      : 0x3ffb1ea0  
A2      : 0x00000000  A3      : 0x00000032  A4      : 0xfffffffe  A5      : 0x3ffc21d4  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffc1dd4  A9      : 0x0000ffff  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffc21d4  
A14     : 0x00000460  A15     : 0x00000462  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffffe  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400dc751:0x3ffb1ea0 0x400dc840:0x3ffb1ee0 0x400d9bae:0x3ffb1f00 0x400ddf9c:0x3ffb1f30 0x400fb646:0x3ffb1f60 0x400fbf76:0x3ffb1f90 0x401190b5:0x3ffb1fb0 0x4008b0a2:0x3ffb1fd0
```

### Anything else?

I have not tried this on a different ESP32, so I'm open to the possibility there's something wonky with this particular board's memory.  If it is that, then forgive me for opening this and please promptly close it.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Aircoookie, 2021-12-05:
#1854 , already fixed since half a year ago. I need to make more frequent releases 😅 please upgrade to 0.13.0-b5, the issue is fixed there.

closed 2021-12-05 by Aircoookie

Shoopity, 2021-12-05:
My apologies.  I've stayed away from the "b" releases since they're "beta".  Of course, nowadays, people throw those terms around so loosely (I'm sorry, but if I can play a level of a game, with full textures and effects... that's not an alpha release), it's hard to know if something is ready for general consumption.

I've read your posts about changing priorities and how this passion project turned into something so amazing that people are putting it into their products to sell.  I swear I searched the issues for "drip" prior to opening this, but apparently I didn't have the search terms correct (I think I was only looking for open issues), and I now see the issue in April.  I promise I'll be more thorough in my searching in the future.