wled/WLED #2387: wifi DMX converter wf310

Imperattor, 2021-12-04:
How to run DMX control via Art-Net wifi dmx converter ? e.g. WF310
please help!

Aircoookie, 2021-12-05:
The WF310 is a wifi->DMX converter. WLED can do the same job with DMX output compiled in and a MAX485 attached to GPIO2, but it makes no practical sense to _combine_ a WF310 with a WLED instance.

closed 2021-12-05 by Aircoookie

Imperattor, 2021-12-05:
> The WF310 is a wifi->DMX converter. WLED can do the same job with DMX output compiled in and a MAX485 attached to GPIO2, but it makes no practical sense to _combine_ a WF310 with a WLED instance.

Thank you. But I'm looking for some way to install and run the DMX control and don't understand how to do it ( 
Can you tell me step by step?
board ESP-01

Imperattor, 2021-12-05:
What software do I need to do this? 
e.g. how to control the wled via resolume ?

Imperattor, 2021-12-05:
ledfx controls wled via wifi
How can I do the same from another light control program?

Imperattor, 2021-12-05:
I need this!
https://www.youtube.com/watch?v=Vrg2UhUGqcc&ab_channel=dBLumsolutionaudiovisuelle

Aircoookie, 2021-12-05:
WLED can receive standard Art-Net or E1.31 data. Just select it in sync settings and enter the WLED device IP address in resolume.

Imperattor, 2021-12-05:
> WLED can receive standard Art-Net or E1.31 data. Just select it in sync settings and enter the WLED device IP address in resolume.

Thank you! I'll be sure to try it and report back.

Imperattor, 2021-12-05:
Can't find anything like that. 
Where i need put the IP address in resolume?
there is no such item in the in DMX tab

Aircoookie, 2021-12-05:
No clue, sorry. All that I can help you with is that WLED can receive Art-Net data, not how to configure a sending software to output it.

Imperattor, 2021-12-05:
> No clue, sorry. All that I can help you with is that WLED can receive Art-Net data, not how to configure a sending software to output it.

OK, I'll try to find a solution