wled/WLED #2389: Segments are not saved

viktor6, 2021-12-05:
### What happened?

Segments are not saved

### To Reproduce Bug

I break the tape into segments
1 segment 0-75
2 segment 75-95
save and overload

### Expected Behavior

To keep segments after restart

### Install Method

Binary from WLED.me

### What version of WLED?

Server message: WLED 0.13.0-b5 (build 2111170)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2021-12-05:
You need to save a preset.
Not a bug.

closed 2021-12-05 by blazoncek

shampeon, 2021-12-05:
This is still a bug, actually. I originally set up my ESP32 with 50 LEDs, and then added more. The Segments only show 50. When I update the segment to 250, click Apply, and reboot, it goes back to a single segment of 50.

shampeon, 2021-12-05:
Before:
[image]
[image]

After reboot:
[image]

blazoncek, 2021-12-05:
Segments are saved in **PRESET** not by themselves.

Aircoookie, 2021-12-05:
@shampeon still a feature 🙂 If you only have a single segment and plan to add more LEDs and want the preset to adjust dynamically, uncheck "Save segment bounds"

shampeon, 2021-12-05:
Ok. Then I'd suggest moving the segment dialog under the presets to make it clear that segments are per preset in the UI. It looks like they're distinct things.

Aircoookie, 2021-12-05:
It's already arranged that way, just left to right: Colors and palettes are part of effects which are part of segments which are part of presets. Putting everything underneath would make the UI very impractical to use.